lumapu/ahoy #1746: Power limit not accepted since v 0.8.34 for HM600

mabaur, 2024-09-12:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.34

### Github Hash

3178325

### Build & Flash Method

was already installed

### Setup

nothing, only Wlan config. Mqtt off. 

### Debug Serial Log output

```shell
0.7.36

08:43:40 I: (#0) resetPayload
08:43:40 I: sendControlPacket cmd: 0x0b
08:43:40 I: TX 13B Ch61 | 51 61 00 02 62 86 16 27 95 5a 5a 64 16 
08:43:40 I: clearCmdQueue
08:43:40 I: (#0) enqueCommand: 0x05
08:43:41 0) Response from devcontrol request received
08:43:41 I: (#0) has accepted power limit set point 100 with PowerLimitControl 0
08:43:41 I: clearCmdQueue
08:43:41 I: (#0) enqueCommand: 0x05

0.8.34 - 3178325
---------------
08:37:40.747 I: (#0) sendControlPacket cmd: 0b
08:37:39.185 -----
08:37:40.685 I: (#0) sendControlPacket cmd: 0b
08:37:41.938 I: (#0) RX  33ms | 12 CH03 | d1 5a
08:37:41.939 W: unknown dev ctrl: 40
08:37:41.940 I: (#0) has not accepted power limit set point 100 with PowerLimitControl 0
08:37:41.941 -----
```

### Error description

Since v0.8.34 ( also observed with 0.8.127 ) PowerLimit can not be set for HM600 inverter. Other tested inverter ( HMS1800 ) works fine. See serial log.

Gubi2023, 2024-09-12:
please use latest release 0.8.140 and check again!

technics42, 2024-09-12:
Using 0.8.101 and power limit can be set without any problems via web interface.

mabaur, 2024-09-13:
Checked with 0.8.140. Same issue: 
@technics42 have you used HM600 inverter ? My HMS1800 also works fine. 
Protocoll structure hase somehow changed from 0.7.37 to 0.8.34  - see serial log above. 
What does "unknown dev ctrl: 40" mean ? 

0.8.140 - f1f4481
---------------

05:59:12.135 I: (#0) sendControlPacket cmd: 0b
05:59:12.387 I: (#0) sendControlPacket cmd: 0b
05:59:12.639 -----
05:59:12.640 I: (#0) sendControlPacket cmd: 0b
05:59:12.722 I: (#0) RX  34ms | 12 CH75 | d1 5a
05:59:12.724 W: unknown dev ctrl: 40
05:59:12.724 I: (#0) has not accepted power limit set point 100.00 with PowerLimitControl 0
05:59:12.726 -----
05:59:12.727 I: (#0) sendControlPacket cmd: 0b
05:59:12.979 I: (#0) sendControlPacket cmd: 0b
05:59:15.105 I: (#0) sendControlPacket cmd: 0b
05:59:15.357 I: (#0) sendControlPacket cmd: 0b
05:59:15.442 I: (#0) RX  38ms | 12 CH03 | d1 5a
05:59:15.443 W: unknown dev ctrl: 40
05:59:15.443 I: (#0) has not accepted power limit set point 100.00 with PowerLimitControl 0
05:59:15.445 -----
05:59:15.446 I: (#0) sendControlPacket cmd: 0b
05:59:15.698 I: (#0) sendControlPacket cmd: 0b
05:59:15.783 I: (#0) RX  37ms | 12 CH40 | d1 5a
05:59:15.784 W: unknown dev ctrl: 40
05:59:15.784 I: (#0) has not accepted power limit set point 100.00 with PowerLimitControl 0
05:59:15.786 -----
05:59:15.786 I: (#0) sendControlPacket cmd: 0b
05:59:15.867 I: (#0) RX  32ms | 12 CH75 | d1 5a
05:59:15.868 W: unknown dev ctrl: 40
05:59:15.869 I: (#0) has not accepted power limit set point 100.00 with PowerLimitControl 0
05:59:15.870 -----
05:59:15.871 I: (#0) sendControlPacket cmd: 0b

Gubi2023, 2024-09-13:
at me all works fine! see screenshot.
[image]

do you get other data of the hm600?

mabaur, 2024-09-13:
Yes, it says limit 0% but gives power ? Strange. 
[image]

Gubi2023, 2024-09-13:
have you tried other values or change W to %  ? maybe your inputs have wrong size?

Gubi2023, 2024-09-13:
something is strange in your picture: you dont have a yield total and no update-time...

Gubi2023, 2024-09-13:
what happens, when you make not persistent? may be the hm cant store

Gubi2023, 2024-09-13:
you can activate the logs in your settings and look what serial debug shows afterwards....

[image]

technics42, 2024-09-13:
Yes, I am using HM600, otherwise I wouldn't have written something in this issue.

mabaur, 2024-09-13:
Yes, who knows :-) Meanwhile I think it is something with my Inverter. I flashed also old versions of Ahoy-DTU on the ESP8266 and all have the same issue. With 0.8.140 I have problems with the ES8266, it often crashed and so I tried it with the Fusion board wich supports differend RF modules for HM600 and HMS1800 based on ESP32- S3. But also here the same :-( .
[image]

knickohr, 2024-09-13:
How do you send the Command ?

It says that your device is wrong, and it is wrong !

mabaur, 2024-09-13:
How I send the command ? I don't understand the question  - but see here: 
[image]
What do you mean with wrong ? Wrong Serial number or not supported as in openDTU ?

Gubi2023, 2024-09-13:
welche version hast du geflasht? wie oben erwähnt macht die -min-Version Probleme beim Limit setzen!

mabaur, 2024-09-13:
@knickohr : Gute Idee, dass was er anzeigt bringt er zwar auch ( Messsteckdose), aber das Befehle werden ignoriert. 
@Gubi2023 : Auf das ESP8266 board zu testen: 0.6.9., 0.7.36, 0.8.34, 0.8.140, 0.8.140-minimal. Auf dem ESP32-S3 openfusion Arbeitsboard 0.8.140. Aber es liegt jetzt mit Sicherheit an am Wechselrichter und nicht an der SW. Warum ist das zuerst dachte siehe log ganz oben. Da wird in 0.7.36 und älter das Limit  bestätigt. Der WR macht aber trotzdem nix.

knickohr, 2024-09-13:
Wenn gar nichts hilft, den WR komplett vom Strom nehmen, auch die PV-Seite und mal e8ne halbe Stunde liegen lassen bis sic/ die Kondensatoren komplett entladen haben.

mabaur, 2024-09-13:
Bis jetzt geht es leider noch nicht. Anschalten ( auch Aus und An, vom 220V Netzt trennen ) hat nicht geholfen. PV-Seite abstecken wird schwierig, die Anlage ist auf dem Dach - schwer zugänglich. Müsste die sich bei Nacht nicht wie "ohne Solarzellen" verhalten ? Also bei Nacht mal eine Weile ausstecken ? .

knickohr, 2024-09-13:
Es müssen beide Seiten aus sein, ansonsten funktioniert es nicht. Dann mußt Du warten bis es richtig Nacht ist und es nochmal probieren.

Möge die dunkle Seite der Macht mit Dir sein !

mabaur, 2024-09-16:
So, habe viel probiert ( ausstecken über Nacht, usw. , aber irgendwie mag der Wechselrichter sich nicht mehr regeln lassen. Die Leistung, die er anzeigt kommt im Netz auch an, d.h. er ist auf 100% eingestellt, auch wenn die Ahoy-DTU GUI was anderes anzeigt.
@Gubi2023 Welche Gen. des HM600 hast DU? Meiner ist nämlich einer der Ersten Gen2 mit  104 am Anfang der Seriennummer.

Ich habe hier noch was gefunden: https://github.com/tbnobody/OpenDTU/issues/863 das alles beantwortet und mache das dann zu.

closed 2024-09-16 by mabaur

rejoe2, 2024-09-17:
Sieht nach einem 2nd gen MI aus (1041, nicht: 1042). Hat vermutlich nie funktioniert, nur die Meldung hat sich irgendwann geändert....

mabaur, 2024-09-19:
@rejoe2 Ja, ist 1041. Hab grad nochmal die Rechnung aus 2020 rausgesucht - ist tatsächlich ein MI600. 
@stefan123t : das weiß ich nicht mehr sicher. Ich habe das mal vor zwei Jahren mit älteren AhoyDTU Versionen angefangen und dann lange liegen lassen. Ich kann auch leider nicht sagen ob es wirklich jemals funktioniert hat, da ich mich damals auf den "falschen" Log verlassen habe und nicht groß kontrolliert habe. Was ja froh wenn dass über Mqtt gezuckt hat . Das mit dem Flash habe ich mir auch überlegt. Aber müsste dann nicht non-persistant trotzdem gehen ? Das wird ja ins RAM geschrieben.

rejoe2, 2024-09-19:
> Ich kann auch leider nicht sagen ob es wirklich jemals funktioniert hat, da ich mich damals auf den "falschen" Log verlassen habe und nicht groß kontrolliert habe.

Kannst ja mal checken, ob mit der alten (oder neuen...) Version  effektiv was passiert an der produzierten Leistung. Afaik geht es mit den 600ern nicht, allenfalls mit MI1500 (oder es braucht irgend eine Initialisierung, die wir (noch) nicht kennen).