lumapu/ahoy #1747: Olimex ESP32-POE-ISO and AhoyDTU

Alpha4Linux, 2024-09-13:
Does AhoyDTU support PoE and Ethernet-LAN functionality on Olimex ESP32-POE-ISO ?
It has a LAN87x0 chip for Ethernet.
Already saw that OpenDTU has Ethernet-LAN and PoE support and I'm curious about AhoyDTU.

Would love to connect the AhoyDTU on PoE and Ethernet-LAN to get rid of external Power Supply and more stable network connection.

appreciate feedback

lumapu, 2024-09-21:
maybe related:
* #886

Alpha4Linux, 2024-11-21:
the last build of ahoy_esp32-wroom32**-ethernet** was build 0.8.133. did it include support for the LAN87x0 chip?

Alpha4Linux, 2025-04-04:
Hi all
is there a chance to have Ethernet (and PoE) support integrated in AhoyDTU in the near future ?
also noticed there's very little activity here

lumapu, 2025-04-04:
for now only W5500 is supported. The activity will increase soon, once the next version is ready for the community.