lumapu/ahoy #1748: Problem beim Update auf 0.8.144

Gubi2023, 2024-09-15:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.144

### Github Hash

97d9786

### Build & Flash Method

was already installed

### Setup

Update von 0.8,134 auf .144

### Error description

Guten Morgen!
Habe heute nacht entdeckt, dass ein neues Update verfügbar ist und es frohgemut installiert. (nachts)
Schien auch zu funktionieren, leider heute morgen Meldung WR nicht erreichbar.. Erst nach einem 2. Update wurde der WR gefunden. Kann das mit der Änderung zu tun haben, dass der WR, wenn er bei Installation nicht da ist (nacht) dann nicht automatisch erkannt wird?

knickohr, 2024-09-15:
In dem Fall würde ich erst mal von der 134 auf die 140 gehen. Jedenfalls in kleinen Schritten.

Gubi2023, 2024-09-15:
tschuldigung, Tippfehler! War von 143 auf 144. Und das Problem hatte ich vorher so nie, da war es egal wann ich upgedatet hatte. (PS: Webserial sagte auch noch Nachtmodus und CMT unknown)

Gubi2023, 2024-09-15:
@lumapu:  Na toll! Jetzt hat nach 4 Std. sich meine MqTT-Übertragung verabschiedet! Ist da vllcht bei der Fehlerbeseitigung MqTT-Überlauf was schiefgelaufen??

Gubi2023, 2024-09-15:
19:Uhr10 .144 nochmals neu installiert--> MqTT läuft wieder. 20 min später wieder aus.   Daraufhin .143 wieder verwendet. Mal sehen....

Gubi2023, 2024-09-17:
neues Verhalten mit 0.8.144: Bisher gab es einen Neustart der DTU, jetzt wird nur noch MqTT beendet, die DTU läuft weiter    
[image]

lumapu, 2024-09-17:
ist mir neu, der Fehler. Habe an MqTT nichts gemacht

Gubi2023, 2024-09-17:
schon klar, muss irgendein Sideeffekt durch die Änderung von 143 auf 144 sein, Fehlerbild ist definitiv wie beschrieben.

knickohr, 2024-09-17:
Bei mir läuft 144 jetzt seit 2 Tagen. Ich kann nichts mysteriöses feststellen 🤔

Gubi2023, 2024-09-17:
Hattest du MqTT-Überläufe? Der Fehler hat m.E. diesen ersetzt...

knickohr, 2024-09-17:
Nein, allerdings mache ich JSON-Payload, eben weil es (früher) mal diese Überläufe gab.

Eigentlich sind es gar keine Überläufe sondern der Broker nimmt einfach nicht schnell genug ab. Vor allem damals als es noch einzelne Topics waren. Und bei 16 Invertern kommt da schon mächtig was rüber 😲

knickohr, 2024-09-19:
Ich hatte eben genau den gleichen MQTT-Ausfall. Es hat einfach nach knapp 2 Tagen aufgehört 😢

[image]

Es hat sich nicht mehr reconnected. Selbst ein manueller Reboot brachte nichts, ich mußte es komplett stromlos machen und neu starten. Ein automatischer Reboot fand auch nicht statt.

lumapu, 2024-09-19:
irgendwas hat das wohl mit der NTP Zeitabfrage zu tun, weiß aktuell noch nicht was. Habt ihr Ideen?

lumapu, 2024-09-19:
ich habe eine Uptime von 4,5 Tagen und bisher keinen MqTT-Ausfall beobachten können

blueline13, 2024-09-19:
Ich hänge mich mal hier mit dran weil ich auch zum Teil betroffen bin.

Ich benutze neuerdings eine DTUFusion Board mit Ethernet(vorher nur WLAN und da passiert es auch) 0.8.144. Bei mir läuft es immer so grob einen 3/4 Tag mit Limit Regelgung (schreiben via API und lesen des Ack via MQTT). Auch bei mir kommt nichts mehr über MQTT zurück. Doch die Kommunikation in Richtung Inverter läuft weiterhin. Bei mir reicht es die DTU über das WEB-IF neu zu starten.

Alternativ läuft auch öfter ein ESP32 (0.8.144) doch der steigt schon deutlicher eher aus. Der bootet nach so 1-2 Stunden. Je nachdem wie viel geregelt werden muss. Das Problem ist schon in #1600 ausführlich beschrieben.

Gubi2023, 2024-09-21:
grade festgestellt, dass unter System auf einmal die Angaben zu Sonnenauf/untergang verschwunden sind! (anscheinend mit Beendigung von MqTT verloren gegangen)
[image]

Gubi2023, 2024-09-21:
die DTU hat tatsächlich die Nachtabschaltung "vergessen", wie aus dem Weblog ersichtlich
[image]

Gubi2023, 2024-09-22:
Mysterium DTU!  über Nacht durchlaufen lassen, heute morgen geht wieder alles richtig! (MqTT, sunrise/set....)

Gubi2023, 2024-09-22:
@lumapu : grad gesehen, dass es ne neue Dev gibt, bin aber noch auf .144. Hast du ein automatisches Update eingebaut?😉

lumapu, 2024-09-22:
nein, nichts automatisches, alles beim Alten (auf Update bezogen)

Gubi2023, 2024-09-22:
hab jetztdie 8.145 installiert. leider nach ca. 4 Std wieder DTU-Reboot, diesmal Grund des Neustarts  "Panic"
@lumapu Coredump im discord

Gubi2023, 2024-10-08:
auch mit der 8.152 reboot nach ca. 9 Std. mit task watchdog
@lumapu Coredump im discord

knickohr, 2024-10-08:
Da gibt es noch einen weiteren (oder ist es der gleiche ?) Bug : 

Wenn ich Nachts ein Update einspiele bootet es am Morgen bei Comminication Start genau einmal. Danach funktioniert es problemlos, auch über mehrere Tage.

Gubi2023, 2024-10-08:
hatte ich eine Zeitlang auch, dass die DTU jeden morgen gegen 8Uhr neu bootete. Da lieferte sie aber schon ca. 1 Std Daten.
Diesmal waren es ca. 2 Std