lumapu/ahoy #1749: Problem with first connection to HMS-1600-4t

cypel21, 2024-09-15:
## Hardware
  * [ ] ESP8266
  * [ x] ESP32
  * [ ] Raspberry Pi

Modelname: ESP32
Retailer URL: https://a.aliexpress.com/_EHoIPTB

### CMT2300A
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip
* [x ] https://pl.aliexpress.com/item/1005005314879396.html

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.140 - f1f4481
Github Hash: _______

### Build & Flash Method:
* [X ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Good morning

This is my first post. Unfortunately I have not been able to connect to my HMS-1600-4t inverters. Yesterday I was able to connect the inverters to the grid.

I added them to the inverters, the radio shows as visible in the settings. Unfortunately I am unable to establish a connection. I have tried to check every frequency and unfortunately with no results.

I put the equipment practically right next to the inverters. The only barrier is the roof tile.

Is there something I can't find?

[image]

0.8.140 - f1f4481
---------------
12:29:58.000 W: CMT TX failed, code: 2
12:29:58.045 W: CMT TX failed, code: 2
12:29:58.046 -----
12:29:58.046 W: CMT TX failed, code: 2
12:29:58.099 W: CMT TX failed, code: 2
12:29:58.099 -----
12:29:58.100 W: CMT TX failed, code: 2
12:29:58.119 W: CMT TX failed, code: 2
12:29:58.119 -----
12:29:58.120 W: CMT TX failed, code: 2
12:29:58.284 W: CMT TX failed, code: 2
12:29:58.284 -----
12:29:58.285 I: com loop duration: 285ms
12:29:58.286 -----
12:30:13.000 W: CMT TX failed, code: 2
12:30:13.136 W: CMT TX failed, code: 2
12:30:13.137 -----
12:30:13.138 W: CMT TX failed, code: 2
12:30:13.189 W: CMT TX failed, code: 2
12:30:13.189 -----
12:30:13.190 W: CMT TX failed, code: 2
12:30:13.231 W: CMT TX failed, code: 2
12:30:13.231 -----
12:30:13.232 W: CMT TX failed, code: 2
12:30:13.329 W: CMT TX failed, code: 2
12:30:13.329 -----
12:30:13.330 I: com loop duration: 330ms
12:30:13.331 -----
12:30:43.000 W: CMT TX failed, code: 2
12:30:43.048 W: CMT TX failed, code: 2
12:30:43.049 -----
12:30:43.049 W: CMT TX failed, code: 2
12:30:43.213 W: CMT TX failed, code: 2
12:30:43.213 -----
12:30:43.214 W: CMT TX failed, code: 2
12:30:43.350 W: CMT TX failed, code: 2
12:30:43.350 -----
12:30:43.351 W: CMT TX failed, code: 2
12:30:43.468 W: CMT TX failed, code: 2
12:30:43.468 -----
12:30:43.469 I: com loop duration: 469ms
12:30:43.470 -----
12:30:58.000 W: CMT TX failed, code: 2
12:30:58.114 W: CMT TX failed, code: 2
12:30:58.115 -----
12:30:58.115 W: CMT TX failed, code: 2
12:30:58.208 W: CMT TX failed, code: 2
12:30:58.208 -----
12:30:58.209 W: CMT TX failed, code: 2
12:30:58.252 W: CMT TX failed, code: 2
12:30:58.252 -----
12:30:58.253 W: CMT TX failed, code: 2
12:30:58.372 W: CMT TX failed, code: 2
12:30:58.373 -----
12:30:58.374 I: com loop duration: 374ms
12:30:58.374 -----
12:31:13.000 W: CMT TX failed, code: 2
12:31:13.045 W: CMT TX failed, code: 2
12:31:13.046 -----
12:31:13.046 W: CMT TX failed, code: 2
12:31:13.065 W: CMT TX failed, code: 2
12:31:13.065 -----
12:31:13.066 W: CMT TX failed, code: 2
12:31:13.105 W: CMT TX failed, code: 2
12:31:13.105 -----
12:31:13.106 W: CMT TX failed, code: 2
12:31:13.348 W: CMT TX failed, code: 2
12:31:13.348 -----
12:31:13.349 I: com loop duration: 349ms
12:31:13.350 -----
12:31:28.000 W: CMT TX failed, code: 2
12:31:28.136 W: CMT TX failed, code: 2
12:31:28.137 -----
12:31:28.137 W: CMT TX failed, code: 2
12:31:28.343 W: CMT TX failed, code: 2
12:31:28.343 -----
12:31:28.344 W: CMT TX failed, code: 2
12:31:28.438 W: CMT TX failed, code: 2
12:31:28.438 -----
12:31:28.439 W: CMT TX failed, code: 2
12:31:28.684 W: CMT TX failed, code: 2
12:31:28.685 -----
12:31:28.686 I: com loop duration: 686ms
12:31:28.686 -----
12:31:43.000 W: CMT TX failed, code: 2
12:31:43.053 W: CMT TX failed, code: 2
12:31:43.054 -----
12:31:43.054 W: CMT TX failed, code: 2
12:31:43.218 W: CMT TX failed, code: 2
12:31:43.219 -----
12:31:43.220 W: CMT TX failed, code: 2
12:31:43.355 W: CMT TX failed, code: 2
12:31:43.355 -----
12:31:43.356 W: CMT TX failed, code: 2
12:31:43.518 W: CMT TX failed, code: 2
12:31:43.518 -----
12:31:43.519 I: com loop duration: 519ms
12:31:43.520 -----
12:31:58.000 W: CMT TX failed, code: 2
12:31:58.136 W: CMT TX failed, code: 2
12:31:58.137 -----
12:31:58.137 W: CMT TX failed, code: 2
12:31:58.195 W: CMT TX failed, code: 2
12:31:58.195 -----
12:31:58.196 W: CMT TX failed, code: 2
12:31:58.219 W: CMT TX failed, code: 2
12:31:58.219 -----
12:31:58.220 W: CMT TX failed, code: 2
12:31:58.506 W: CMT TX failed, code: 2
12:31:58.506 -----
12:31:58.507 I: com loop duration: 507ms
12:31:58.508 -----
12:32:13.000 W: CMT TX failed, code: 2
12:32:13.136 W: CMT TX failed, code: 2
12:32:13.137 -----
12:32:13.138 W: CMT TX failed, code: 2
12:32:13.334 W: CMT TX failed, code: 2
12:32:13.334 -----
12:32:13.335 W: CMT TX failed, code: 2
12:32:13.362 W: CMT TX failed, code: 2
12:32:13.362 -----
12:32:13.363 W: CMT TX failed, code: 2
12:32:13.403 W: CMT TX failed, code: 2
12:32:13.403 -----
12:32:13.404 I: com loop duration: 404ms
12:32:13.405 -----
12:32:28.000 W: CMT TX failed, code: 2
12:32:28.136 W: CMT TX failed, code: 2
12:32:28.137 -----
12:32:28.137 W: CMT TX failed, code: 2
12:32:28.423 W: CMT TX failed, code: 2
12:32:28.423 -----
12:32:28.424 W: CMT TX failed, code: 2
12:32:28.560 W: CMT TX failed, code: 2
12:32:28.560 -----
12:32:28.561 W: CMT TX failed, code: 2
12:32:28.790 W: CMT TX failed, code: 2
12:32:28.790 -----
12:32:28.791 I: com loop duration: 791ms
12:32:28.792 -----
12:32:43.001 W: CMT TX failed, code: 2
12:32:43.136 W: CMT TX failed, code: 2
12:32:43.137 -----
12:32:43.137 W: CMT TX failed, code: 2
12:32:43.423 W: CMT TX failed, code: 2
12:32:43.423 -----
12:32:43.424 W: CMT TX failed, code: 2
12:32:43.560 W: CMT TX failed, code: 2
12:32:43.560 -----
12:32:43.561 W: CMT TX failed, code: 2
12:32:43.847 W: CMT TX failed, code: 2
12:32:43.848 -----
12:32:43.849 I: com loop duration: 849ms
12:32:43.849 -----

TheSCC1992, 2024-09-15:
yes, same for me, i went back to the last stable.

Gubi2023, 2024-09-15:
are you shure you have HMS-1600-4t without intern WIFI? this one dont work with Ahoy!

cypel21, 2024-09-15:
> are you shure you have HMS-1600-4t without intern WIFI? this one dont work with Ahoy!

Yes, I am sure

Gubi2023, 2024-09-15:
do you have an entry in marked field?
[image]

cypel21, 2024-09-15:
[image]
@Gubi2023 just like yours

DanielR92, 2024-09-15:
Please send us some additionall information. 
Self builded Hardware, or buyed?
What is the serial number of your inverter?
What pin-configuration do you have now?

TheSCC1992, 2024-09-15:
> > are you shure you have HMS-1600-4t without intern WIFI? this one dont work with Ahoy!
> 
> @Gubi2023 Yes, I am sure
> 
> @TheSCC1992 Did it worked? I used also opendtu and nothing

yes under 0.8.83 it works fine, but when i upgrades to the new stable no communication was available. i waited a whole day but nothing happend. 
Self builded Hardware, or buyed? -> self made but with a bught board
What is the serial number of your inverter? 116492230980
What pin-configuration do you have now? 
[image]

cypel21, 2024-09-15:
@DanielR92 
1) Self build from ready boards (links in the beginning of this issue)
2) 1164A00D4A83
1164A00BECFF
3) Pin below
[image]

lumapu, 2024-09-21:
> `W: CMT TX failed, code: 2`

for me it looks like having a connection problem to the CMT2300A module. How was it assembled / soldered? I have seen the linked AliExpress item, but how is it connected to the ESP? Are you using jump wires? Has anybody else already tested this baord?

TheSCC1992, 2024-09-22:
I don´t think it is a problem with wiring or module. I have got the same problems with my Module. Under  0.83 it is fine, but after updating it does´t work, so there couldn't be a module or wiring problem.

lumapu, 2024-09-22:
Ok, understand, thanks for clarifying. 
Have you tried to install the newer firmware during night or let it at least installed over night? We have seen communication issues while installing different versions during production of the inverters.

TheSCC1992, 2024-09-22:
i installed now on a new esp32 the 0.8.145 version. after booting I get now my HM Series working, but the HMS are offline, i will check tomorrow, if i get some data.

MyWebstage, 2024-09-23:
Ich habe bei den Großeltern ein Balkonkraftwerk mit dem HMS-1600 am Freitag in Betrieb nehmen wollen. Ich hatte noch eine ältere Version der ahoyDTU installiert und durch die Buchstaben in der Seriennummer festgestellt, dass ich updated muss. Nach dem Update auf 0.8.140 kam keine Verbindung Zustand. Bei mir wird in Version 0.8.140 unter System > CMT2300A > "Interrupt Pin funktioniert = False" angezeigt und ich vermutete einen Hardware-Fehler. Bin also wieder nach Hause.

Als ich das hier gelesen habe, habe ich nun die 0.8.83 installiert. Jetzt steht bei "Interrupt Pin funktioniert = Ja".

Vielleicht hilft das bei der Fehlersuche.

Was anderes zum Thema Limiter:
Wir wissen ja, dass wenn man den WR "normal" auf 800W drosselt, alle 4 Inverter auf fest 200W gedrosselt werden. Ist das Active Power Limiter von Ahoy eine dynamische Regelung der Ausgangsleistung auf 800W? Falls ja - erstmal richtig nice! Aber dann könnte man dieses Feature ein bisschen offensichtlicher präsentieren. Falls nein, gibt es dazu bereits eine Feature-Request?

MyWebstage, 2024-09-23:
So ich muss zurück rudern. Auch mit Version 0.8.140 bekomme ich nun Daten und der Interrupt-Pin ist auch wieder grün. Ich habe die Frequenz auf 865MHz geändert und ca. 20 Minuten gewartet.

Elko mit 100yF ist an den Spannungspins des CMT2300A.

Super, also die Software tut. 👍

TheSCC1992, 2024-09-23:
so habe nun einen neuen CMT2300A bestellt, mal schauen ob das läuft

lumapu, 2024-09-23:
ich sage es nochmal: installiert die neue Version am Abend / Nacht und wartet bis zum nächsten Morgen, dann wird alles wie gewohnt funktionieren.