lumapu/ahoy #1750: Gesamtertrag von einem Tag auf den Nächsten vom HM350 reduziert

kritisch1, 2024-09-17:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: _HM350_____
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version:0.8.83
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo,
bei meinem HM350 wurde der Gesamtertrag von einem Tag auf den Nächsten von 283177 Wh auf 218772 Wh reduziert. Der HM350 ist jetzt seit einem Jahr in Betrieb und hat den Gesamtertrag bisher immer kontinuierlich aufaddiert. Das Update von Ahoy auf Version:0.8.83 ist schon einige Monate her. Gibt es für dieses merkwürdige Verhalten eine Erklärung?

Alpha4Linux, 2024-09-17:
Hi
Das habe ich bereits mehrere Male am HM-600 beobachtet. Ich logge den Total Yield in einer DB.
Wie zum Beispiel hier vom 16.8 -> 17.8 und vom 18.8 -> 19.8:

2024-08-16: **131.77 kWh**
2024-08-17: **68.51 kWh**
2024-08-18: **69.82 kWh**
2024-08-19: **6.93 kWh**

Die Ertragsdaten vom Hoymiles messe ich zur Weiterverarbeitung ab einem separaten "Einspeisezähler" aber aufgefallen ist es mir trotzdem. Deshalb auch Interesse an einer Erklärung.
Der HM-600 wird ab LFP-Batterie gespiesen und permanent mit DC am Eingang versorgt, somit kein Unterbruch.

technics42, 2024-09-18:
What about the yields of the inverter inputs? Did they also decrease? Did you enter any offsets for the inputs?

Alpha4Linux, 2024-09-18:
i did not give attention to the DC Inputs Yields so far, but currently their totals do match with the Output Yield kWh.
No "Yield Correction" in use = 0

lumapu, 2024-09-21:
I think it is related with their (Hoymiles) `yield day` counter. After a few days / weeks it overflows because it is only a 16bit value which counts watt-hours (Wh). After `65535Wh` (= `65kWh`) it overflows to `0`. Internally they do the calculation of `yield total` may using the `yield day` counter.
For me it also matches with the values provided by @Alpha4Linux 

Consider to restart your inverter from time to time.

I'm currious if the `yield total` is correct after rebooting the inverter.

knickohr, 2024-09-22:
Moment, langsam. Auch für Dumme !

- Der Inverter ist rund um die Uhr gelaufen ?
- Wir sprechen von YieldDay oder YieldTotal EINES Inverters ?
- Es ist nur ein Inverter ?

Alpha4Linux, 2024-09-22:
Hi
yes, make totally sense if Yield Day counters are 16bits in Wh. Will have a look at the next overflow in detail.
Since i run an HM-800 with both inputs connected to battery it take's double the time for the overflow to happen compared to single-input inverter with the same amount of energy feed to the grid.
Thanks lumapu so far

Alpha4Linux, 2024-10-14:
update on 16bit counter overflow.
Yield Total from Input "DC-IN 1" started from 0 after reaching it's max value (65536), resulting in a updated (lower) Yield Total of the HM inverter. My Hoymiles is running in a DC-coupled setup with battery, counters usually never get a (daily)-reset.
To get reliable metering of Energy fed to the grid, i do now rely on a external DIN-rail mounted Energy meter.
[image]