lumapu/ahoy #1751: Gibt es hier Nutzer mit sterbenden NRF24L01-Modulen?

SvenLuebke, 2024-09-17:
Hallo Leute!

Hattet Ihr auch schon NRF24L01-Module, die ein paar Wochen/Monate funktioniert haben und dann "gestorben" sind?
Vor ein paar Wochen hatte ich nach vielen Monaten das erste Modul, was lt. Ahoy nichts mehr gesagt hat. Nun scheine ich inzwischen schon das zweite Modul zu haben, was nichts mehr von sich gibt?! Wenn das kein Regelfall ist, könnte das an einer zu großen Spannung liegen, oder? Gemessen hab ich noch nicht.

## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [X] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

knickohr, 2024-09-17:
Bitte mal etwas spezifizieren. Es gibt hunderte verschiedener Module von mindestens ebensoviele Anbietern. Generell kann man sagen das Original Nordic-Bausteine nicht sterben.

Gubi2023, 2024-09-17:
mir ging mal eines kaputt, aber da hatte ich die DTU zerlegt und es vermutl. durch statische Aufladung zerschossen.

dtuuser, 2024-09-17:
Bei mir immer noch das erste Modul.
Aber, bei einer 4er Bestellung fehlte auf einem Modul mal ein SMD Bauteil. Das lief gar nicht.

Edibo63, 2024-09-19:
Kaputt ist noch keins gegangen. Habe trotzdem auf die E01-ML01DP5 von Ebyte umgebaut.
Der Verbindung ist um einiges besser geworden.

SvenLuebke, 2024-09-28:
Vielen Dank erstmal für Eure Antworten. Ich hatte wenig Zeit und es gab Wichtigeres zu tun. Heute hab ich mich der Sache mal wieder angenommen und bin etwas verblüfft:

Natürlich hatte ich das ESP-Modul samt NRF-Modul diverse Male vom Strom getrennt, längere Zeit gewartet und dann wieder angesteckt. Außerdem andere Updates geflashed, /reboot aufgerufen usw. . Das alles hat nicht geholfen, um wieder eine Information vom Wechselrichter zu bekommen. Nachdem ich das Ticket hier erstellt hatte, bin ich nochmal in mich gegangen und hatte schon vermutet, dass ich das Modul - aus Versehen - vielleicht an die 5V-Leitung angeklemmt hatte und wollte das erst einmal überprüfen, bevor ich hier weiter schreibe.

Mit einem Multimeter habe ich heute die Spannung nachgemessen: 3.2V, direkt am Modul, sollte also passen. Für diese Messung hab ich auch ein komplett anderes Netzteil genommen (den ESP+NRF damit komplett vom Strom getrennt), wieder angeschlossen und wieder wurden mir keine Daten angezeigt. Also identisches Verhalten wie vor ein paar Monaten, als das erste Modul nichts mehr geliefert hat. Ich dachte mir "OK, somit ist definitiv das zweite Modul defekt". Augen zu und durch, hier liegen noch ein paar Module herum, also das dritte Modul angesteckt und biiing: Die LED wurde blau. "Also die anderen beiden Module sind wirklich defekt." Nochmal zum Testen zurückgesteckt und biiiing: blaue LED...mit dem gleichen Modul, was seit Wochen nicht mehr läuft. Sicherheitshalber hab ich auch nochmal das alte Modul getestet und das läuft nun auch wieder.

Ich versteh es nicht! Könnte höchstens sein, dass
* an den Kabeln einen Wackelkontakt vorhanden ist, aber eigentlich hab ich da genug herumprobiert und nichts gefunden
* sich die Software irgendwo Kaltstart-fest verhakt und mit einem neuen Modul dort herauskommt...ok...das erscheint mir ziemlich unwahrscheinlich zu sein, da die Webseite immer angezeigt wurde

Naja, das Ende vom Lied ist wohl: Die Module sterben doch nicht!

knickohr, 2024-09-28:
Kalte Lötstellen, Steckkontakt, schlecht gecrimpte Kabel 🫣

closed 2024-10-02 by lumapu