lumapu/ahoy #1752: Uhrzeit im Webserial und inm Webinterface unterschiedlich

Ollipop030, 2024-09-19:
### Platform

ESP32

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.144

### Github Hash

abc123

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

kein MQTT, Intervall 5 Sekunden

### Error description

Heute morgen habe ich mit gewundert, warum die WR noch nicht online sind, obwohl am Zähler schon Produktion zu sehen war. In Ahoy ist mir dann aufgefallen, das lt. Webserial noch Nighttime aktiv ist.

`05:51:14.000 I: (#0) no communication to the inverter (night time)`

Die Uhrzeit hier passt aber nicht, es ist schon zwei Stunden später:

[image]

Nach einem Reboot läuft dann auch wieder die Kommunikation, allerdings steht im Webserial immer noch eine falsche Uhrzeit.

Dirk-Ahoy, 2024-09-19:
Hatte ich heute morgen bei einer DTU auch - nach Neustart lief sie dann - ebenfall 144er Version

Dann bei beiden Wechselrichtern  CODE 2 - Time Calibration  15 Stück  - seit dem läuft es aber.   Hatte aber nicht drüber nachgedacht mal im Log zu schauen vorher.   Müssen wir wohl mal im Auge behalten.

Gubi2023, 2024-09-19:
ist ein alter Hut, war schon immer so, dass nach einem Neustart der DTU die Uhrzeit im webserial falsch ist. Sommerzeit -2, Winter -1Std.  Mit refresh wir sie aktualisiert.

Gubi2023, 2024-09-19:
zu 1. Pkt. siehe auch #1748

Ollipop030, 2024-09-19:
> ist ein alter Hut, war schon immer so, dass nach einem Neustart der DTU die Uhrzeit im webserial falsch ist. Sommerzeit -2, Winter -1Std. Mit refresh wird sie aktualisiert.

Trotzdem sollte die Kalkulation der Nighttime aber die reale Zeit berücksichtigen.

zappoXX, 2024-09-20:
Bei mir auch Morgens Offline mit 0.8.144... zurück zur 0.8.143 und WR ist Morgens Online.

Dirk-Ahoy, 2024-09-20:
Bei 2 weiteren DTUs hatte ich heute morgen auch das Problem das das Webserial noch nachts gezeigt hat  - manuelles 2x Sync in den Einstellungen hat sie dann aufwachen lassen - scheint wohl erst nach einigen Tagen Uptime aufzutreten.   Interessant ist das die Zeiten ja passen - muss irgendwo intern was verloren gehen.

Gubi2023, 2024-09-21:
siehe auch https://github.com/lumapu/ahoy/issues/1748#issuecomment-2365286015https://github.com/lumapu/ahoy/issues/1748#issuecomment-2365286015

lumapu, 2024-09-29:
ist das Thema noch aktuell, oder wurde es durch die Version 145 gefixt?

Ollipop030, 2024-09-30:
> ist das Thema noch aktuell, oder wurde es durch die Version 145 gefixt?

Konnte ich nicht testen, ab der Version .144 ist der ESP32 extrem träge, Webinterface funktioniert bei mir auch nicht.

lumapu, 2024-10-01:
der neue Webserver ist ab der `0.8.141` im Einsatz

Ollipop030, 2024-10-02:
Die .149 läuft, danke dafür. Vielleicht war ich auch von dem Fehler betroffen wie in #1757 beschrieben? Ich sende alle paar Sekunden per API das Limit neu, hatte vergessen den Dienst zu beenden.

Morgen früh scheue ich mal nach der Uhrzeit.

Ollipop030, 2024-10-03:
Das Problem mit der Uhrzeit ist bei der .149 nicht mehr vorhanden.