lumapu/ahoy #1753: Support for HERF-300 1ch inverter

xenio, 2024-09-21:
Is it possible to add the support for the HEF-300 1CH Inverter?

OpenDTU recently added support for this micro inverter 

Feature Request:
https://github.com/tbnobody/OpenDTU/issues/2236

Implementation:
https://github.com/tbnobody/OpenDTU/commit/a94977696645eab161bd3c8445c2ea7213c99ef5

lumapu, 2024-09-21:
I checked to add the serial which is mentioned in the issue and it works like a charm.
Which problem do you face with this type of inverter?

xenio, 2024-09-22:
My bad, I had flashed an old Ahoy firmware.
The latest version recognises the HERF 300, I'll do some testing.
Thanks. 👍

stefan123t, 2025-05-23:
@xenio / Sandro I have just committed PR #1835 to fix #1829 for the HERF_2CH.cpp byte order. I have also updated the HERF_1CH.cpp according to the OpenDTU implementation.
Could you kindly build the PR and try it with your HERF 300 to check if byte order / value assignments are correct for you too ?