lumapu/ahoy #1754: Reboot at Noon

Peertje-62, 2024-09-23:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.145

### Github Hash

2a99097

### Build & Flash Method

ESP Tools (flash)

### Setup

Hm 600 & Hm 300
Openfusion DTU
MQTT

Hm 400
Openfusion DTU
MQTT

### Debug Serial Log output

```shell
-
```

### Error description

Unwanted reboot at 12:00, on my system and also on my neighbor's
Kind regards

knickohr, 2024-09-23:
Jepp, und von einem weiteren User und von mir bestätigt. Liegt vermutlich an einem Bug in „ Reboot Ahoy at midnight“.

hrolofs, 2024-09-23:
same for me

Ollipop030, 2024-09-23:
Wie läuft denn die .145 generell bei euch? Auf meinem ESP32 ist das Webinterface sehr träge, auch nach einem Reboot. Die Nulleinspeisung über REST API hat auch viele Aussetzer, MQTT ist aus.
Die .144 läuftf deutlich besser, zumindest bei mir.

Gubi2023, 2024-09-23:
bei mir ist die Aktualisierung an I-Phone sehr träge...

closed 2024-09-23 by Peertje-62