lumapu/ahoy #1755: 184, 185 und 186 Problem beim Login

kiu77, 2024-09-25:
## Hardware
  * [ ] ESP32
 ### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 

Problem:
Bei diesen Firmware-Versionen kommt nach jedem Klick auf Login auch schön das Login-Fenster,
doch hat man das PW da eingegeben und auf Login geklickt, wird das Eingabefeld einfach gelöscht.
Man ist dann zwar eingeloggt, wird aber nicht mehr auch die Übersichtsseite zurückgeroutet,
was irgendwie nicht schön ist. Das abgebildete Fenster inkl. Link mit /login im Browser bleibt.
[image]

knickohr, 2024-09-25:
144, 145 und 146 !

RealNBB, 2024-09-26:
ja ist "eigentlich" mit der 146 gefixt worden:

fix redirect after login

hab das Problem aber auch weiterhin mit der 146.

kiu77, 2024-09-28:
> 144, 145 und 146 !

Ja, peinlich, danke. Habs korrigiert...

RealNBB, 2024-10-02:
Hi,

mit der 0.8.149 funktioniert alles wieder wie es soll, zumindest bei mir.

kiu77, 2024-10-03:
Kann ich bestätigen: Bei mir auch!

closed 2024-10-03 by kiu77