lumapu/ahoy #1756: NTP timeserver reachable

RealNBB, 2024-09-28:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.146

### Github Hash

83b386d

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Nothing changed

### Error description

Hi,

seit der 1.46 habe ich Probleme mit der Zeitsynchronisation. 
In diesem Fall gab es einen automatischen Restart um Mitternacht, dann das Problem mit der Synchronisation.

Grüße

lumapu, 2024-09-29:
Hat Ahoy zu dem Zeitpunkt Netzwerkzugriff? Besteht Internetzugriff? Ist automatischer Neustart mitternachts aktiviert?

RealNBB, 2024-09-29:
Hi, 3x ja!

closed 2024-10-01 by RealNBB