lumapu/ahoy #1761: Udate to 0.8.151 reports succefull but Version shows 0.8.145

Jawada74, 2024-10-06:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.145

### Github Hash

2a99097

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverter intervalls to 12 Sec.

### Error description

After updating from 0.8.145 to the 0.8.151 the DTU reports succefull update but shows 0.8.145.

RealNBB, 2024-10-06:
Then try again! ;)

If it still doesn't work, make a backup of the config and reinstall it using Webflasher ("erase"). Then import the config and you're done.
It happens from time to time that Ahoy doesn't accept the new version.

Gubi2023, 2024-10-06:
sometimes is usefull to refresh, then the actual version is visible

Jawada74, 2024-10-06:
Refresh had no effect.

[image]

Gubi2023, 2024-10-06:
du hast nicht upgedatet! sonst wäre die uptime auf 0 Tage.

Dirk-Ahoy, 2024-10-06:
einfach mal einen reboot machen dann sollte die neue version laufen

knickohr, 2024-10-06:
Im AP booten 7nd dann den Update machen. Hat bis jetzt immer funktioniert 😉

Jawada74, 2024-10-06:
Seh ich auch so aber ich kann nur sagen das Update Läuft durch und ich bekomme die meldung Update successfull.  und das DTU bootet neu. Danach ist aber alles beim alten.
Ich schaue mal wie ich das AP booten hin bekomme.
Danke

knickohr, 2024-10-06:
Wenn der Update mit Success durchläuft dann einfach manuell booten, also ausschalten und wieder Strom einschalten. Sollte genügen. Nicht auf den automatischen Reboot warten, aber unbedingt auf das Success warten !

Jawada74, 2024-10-06:
Danke Knickohr genau das war es !!!
Einfach Stromlos gemacht und dann wieder angesteckt und schon war die 0.8.151 im display und im webGui.
Hätte ich auch alleine drauf kommen können.

closed 2024-10-06 by Jawada74