lumapu/ahoy #1762: [Bug] 0.8.151 Compile fails with fresh installed PlatformIO

fsck-block, 2024-10-06:
Compile of ahoy fails when compiling with a fresh installed PlatformIO environment.
```
Resolving esp8266-minimal dependencies...
Library Manager: Installing git+https://github.com/nRF24/RF24.git#v1.4.8
git version 2.43.0
Klone nach '/home/XXXXX/.platformio/.cache/tmp/pkg-installing-r9s9ze6y' …

...

Library Manager: ESPAsyncWebServer-esphome@3.2.2+sha.0873c38 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing esphome/ESPAsyncTCP-esphome
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESPAsyncTCP-esphome@2.0.0 has been installed!
Library Manager: Installing git+https://github.com/me-no-dev/ESPAsyncUDP
git version 2.43.0
Klone nach '/home/XXXXX/.platformio/.cache/tmp/pkg-installing-jpyd70c7' …

Library Manager: ESPAsyncUDP@0.0.0+sha.697c75a has been installed!
Updating metadata for the vscode IDE...
UserSideException: Processing esp8266-minimal (platform: espressif8266; board: esp12e; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
ModuleNotFoundError: No module named 'pkg_resources':
  File "/home/XXXXX/.platformio/penv/lib/python3.12/site-packages/platformio/builder/main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "/home/XXXXX/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/XXXXX/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/XXXXX/ahoy-development03/scripts/auto_firmware_version.py", line 5:
    import pkg_resources
========================== [FAILED] Took 0.48 seconds ==========================

Environment      Status    Duration
---------------  --------  ------------
esp8266-minimal  FAILED    00:00:00.478
==================== 1 failed, 0 succeeded in 00:00:00.478 ====================
```
Current PlatformIO is based on python3.12 where the package pkg_resources is no longer available by default.
pkg_resource is used in script auto_firmware_version.py.
Bugfix available. See PR #1759

lumapu, 2024-10-07:
Thank you for pointing that out. I don't understand why the GitHub actions never failed at this point.
Will merge the PR asap.

Gubi2023, 2025-03-09:
release  0.8.154 is buggy, and no more declared as release