lumapu/ahoy #1763: Web Konsole,    webSerial, buffer overflow!

Andy811963, 2024-10-07:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

board

### Connection picture

- [X] I will attach/upload an image of my wiring

### Version

08.151

### Github Hash

e601bfe443342d1b7c85913b1852cac50c4c5475

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nichts.

### Error description

Wenn ich die Web Konsole öffne, dann gibt es einen, buffer overflow!,  bei alten Versionen nicht.
Das gilt für Edge, FireFox, Brave.
Wenn ich dann zurück will startet die Software neu.
[image]
[image]

lumapu, 2024-10-07:
wie viele Wechselrichter werden abgefragt? 
Eigentlich sollte dort kein overflow stattfinden, zumal die ausgegebenen Infos überschaubar aussehen.

Andy811963, 2024-10-07:
Nur 1 HM-1500 mit 4 Modulen a 385 Watt.
Bei alten Versionen hat es noch funtioniert.
Sonst geht aber alles.
Gruß Andy B.

Gubi2023, 2024-10-07:
ich würd mal mit erease neu flashen und schauen, obs besser ist...

Andy811963, 2024-10-07:
Bei Version 8.8.3 geht alles.
[image]

Gubi2023, 2024-10-07:
dann mach mal OTA-Update über https://fw.ahoydtu.de/tools/webflasher/

Andy811963, 2024-10-07:
So,
nach einigen Treiberprobleme mit win 11 ist nun alles geflasht.
Leider immer noch keine Lösung.
[image]

knickohr, 2024-10-07:
Das Webseriel, bzw. Das Debug etwas schlanker machen, sprich die Ausgaben reduzieren. Man braucht nicht immer den kompletten Debug, schon gar nicht im normalen Betrieb.

[image]

Das reicht, zumal Du noch den ESP8266 benutzt.

Andy811963, 2024-10-07:
Hallo,

 Das funktioniert schon besser.

 Aber wenn ich wieder zurück gehe ist alles weg.

 Ich verstehe nicht warum eine ältere Version funktioniert.

 Gruß Andy B.
--Gesendet mit der GMX Mail App

  

  IMG_3044.jpeg (view on web)