lumapu/ahoy #1765: Ahoy DTU verliert permanent WLAN Verbindung bei TP Link Repeater

CLAM01, 2024-10-09:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.140

### Github Hash

f1f4481

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

2 WR Hm600 und HM300
Mqtt alle 10 Sekunden an HA Server
FB als NTP
Feste IP inkl. Gateway

### Error description

Die DTU wird seit neuestem an einem TP-Link Repeater RE190 betrieben.
Diese hat sowohl 2,4 und 5Ghz aktiviert.
Seit Einsatz dieses Repeaters verliert die DTU nach kurzer Zeit seine Wlan Verbindung und connected auch nicht neu.
Erst nach einem reset über den Reset Button kommt die DTU wieder hoch.

Probeweiser wurde das 5Ghz abgeschaltet aber keine Besserung.

Weitere Schritte die unternommen wurden sind, dass die Firmware auch die 0.8.36 runtergezogen wurde.
Auch hier das gleiche Verhalten.

Danach einstellen einer festen IP,auch keine Besserung.

Bei der zuvor eingesetzten FB Fritz!600 Repeater waren die Ausfalle nicht in diesem Ausmaß zu erkennen.
Aufgrund Alexa Multiroom bin ich nun auf daß 5Ghz angewiesen, da sonst mein WLAN im 2,4Ghz Bereich in die Knie geht weil amazon music echt ein Datenfresser ist. 

Ich bin ratlos, weil ich keine logs lesen kann.
Der interne AccessPoint der DTU wurde in diesem Fall nicht aufgespannt.

Was kann ich tun?

knickohr, 2024-10-09:
Dein TP-Link wird wohl,so etwas wie bandsteering und/oder channelsteering haben. Deaktiviere das mal.

CLAM01, 2024-10-09:
Habe so eine Einstellung noch nicht gesehen. Alle meinen anderen esp8266 betriebenen Geräte wie solarnachführung etc. haben kein Problem mit dem TP Link.

CLAM01, 2024-10-09:
Ok, bei der FB ist das automatische umschalten zwischen 2,4 und 5Ghz abgeschaltet weil ich immer zwei SSID habe. Auch bei dem TP Link so.

CLAM01, 2024-10-09:
Ok, ein Test läuft noch.
Auf dem TP Link habe ich nun beide Netze aktiv als repeating, aber stopfe beide Frequenzen in das 5Ghz der FB zur Übertragung an die FB.
Bisher stabil.

FB <5GHz> -- > TP Link <2,4 GHz und 5GHz> --> Endgeräte mit 2,4 und 5 GHz

CLAM01, 2024-10-10:
Über Nacht lief die DTU ohne Probleme.
Sobald die DTU anfängt zu produzieren, schmiert sie nur noch ab und ist nicht mehr erreichbar.
Somit sehe ich nicht den Repeater als Bösewicht.

stefan123t, 2024-10-10:
Das könnte an 802.11r/k/v liegen, also Roaming und/oder Mesh Steering: https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2289412118

Actually both 802.11k and 802.11r are necessary for seamless Basic Service Set (BSS) transitions aka roaming:

802.11k Basic Service Set (BSS) transitions: [IEEE 802.11k-2008](https://en.wikipedia.org/wiki/IEEE_802.11k-2008)
802.11r Fast BSS Transition (FT): [IEEE 802.11r-2008 or fast BSS transition (FT)](https://en.wikipedia.org/wiki/IEEE_802.11r-2008)
802.11v Wireless Network Management: [IEEE 802.11v-2011](https://en.wikipedia.org/wiki/IEEE_802.11v-2011)

und https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2324352636

Did you look into the ESP-IDF example code for 802.11k/r/v mesh steering?

https://github.com/espressif/esp-idf/tree/master/examples/wifi/roaming

But on another thought maybe the events and processes of switching the STA from one of these BSSIDs to another BSSID allow us to determine which event may be missing / needs to be changed to make it work in Arduino/OpenDTU too ?

https://github.com/espressif/esp-idf/blob/6673376297b921d438790d195014b860eaf8bb17/examples/wifi/roaming/roaming_11kvr/main/roaming_example.c#L48-L55

Here they explicitly distinguish the disconnect reason `WIFI_REASON_ROAMING`:

```C++
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        wifi_event_sta_disconnected_t *disconn = event_data;
        ESP_LOGI(TAG, "station got disconnected reason=%d", disconn->reason);
        if (disconn->reason == WIFI_REASON_ROAMING) {
            ESP_LOGI(TAG, "station roaming, do nothing");
        } else {
            esp_wifi_connect();
        }
```

CLAM01, 2024-10-10:
Auch die 0.8.36 das selbe Problem, das sporadische Neustarts auftreten mit "Restart Reason: Exeption".
Ok, nur welche Exeption tritt denn da auf? Wie kann ich das herausfinden?
Also ist die DTU in einem gemischten Netzwerk nicht einsetzbar.
Mit Mesh nur auf 2,4GHz über Fritz!600 geht alles sauber.

Gubi2023, 2024-10-10:
m.W. gibt es für Laien keine Möglichkeit, den Restart-Grund zu entschlüsseln. Bem 8.140-Release gibt es **für ESP32** die Möglichkeit, einen Coredump zu laden, (unter "System") der von @lumapu entschlüsselt werden kann

CLAM01, 2024-10-10:
Also bisher konnte ich feststellen, dass anscheinend die AHOY DTU Probleme bekommt wenn ein Access-Point im 2,4 und 5GHz Bereich arbeitet. Wobei beide SSIDs unterschiedlich sind, sodass kein automatischer Wechsel stattfinden kann zwischen den Frequenzen.

Ich habe nun wieder die 0.8.36 drauf und 5GHz abgeschaltet an dem Access-Point und habe bisher wieder Ruhe.
Ggf. kann sich dem Problem ja mal angenommen werden, denn 5GHz wird immer wichtiger.

Gerne liefere ich dazu was ich nur kann.

knickohr, 2024-10-10:
Das Problem ist nicht die Software, sondern eher die Hardware. So langsam sollten 5GHz ESPs gängig werden. Soweit ich weiß kann es momentan nur der ESP32-C5. Tja, und dann muß natürlich auch ein Stück Software dazu geschrieben werden.

CLAM01, 2024-10-10:
Ja, gebe ich dir recht, gerne hätte ich mehr auf 5Ghz.
Aber es muss trotzdem einen zusammenhang geben zwischen ESP8266 und TP-Link 5Ghz. Ich habe nun bei der FB das 2,4 Ghz und 5 Ghz an und jetzt fühlt sich die Ahoy DTU nicht mehr gestört.

Es kann natürlich auch sein, dass der TP-Link einfach Sch+++e ist und sich selbst auch mal durcheinander kommt, sodass die DTU meint sie wäre getrennt oder sonst was und dann in einen Außnahmezustand "getrieben" wird.

Aus dem Grund hätte ich das gerne mal herausgefunden.

Kann ich das mit dem Seriellen Debuggen herausfinden ggf? Also wenn ich einen Laptop nehme und da die DTU per USB versorge und im Webinstaller die Konsole anschalte?

CLAM01, 2024-10-10:
Jetzt habe ich wieder das 2,4 und 5GHz am TP Link an und die DTU läuft.
So wie gestern.
Es steht also im Zusammenhang damit, dass die beiden WR produzieren. Erst dann fangen die Probleme an.
Hat keiner eine Idee mehr? 

Das ist der einzige unterschied.

CLAM01, 2024-10-11:
Hier der Stacktrace:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffec70 end: 3fffffb0 offset: 0150
3fffedc0:  40262adf 3ffefb00 00000002 40262a74  
3fffedd0:  00000002 40262a1b 00000002 40261b70  
3fffede0:  40261b99 3fffee90 3ffefb00 0000001a  
3fffedf0:  4025f5b5 3fffee90 3ffef9a8 3ffef334  
3fffee00:  3ffed840 4025f5aa 3fffee90 3fff39cc  
3fffee10:  5452414c 31303030 00000000 40101094  
3fffee20:  00006e78 00000dcf 3ffe85f8 40100d12  
3fffee30:  4025b955 00000010 00000020 3fff39cc  
3fffee40:  ffffffcc 3ffef4bc 3ffed850 3ffefb00  
3fffee50:  3ffeea28 0000002c 00000000 402602f7  
3fffee60:  00000000 3fff5234 3ffed840 00000000  
3fffee70:  3ffefb00 00000000 3ffef4e8 ffffffff  
3fffee80:  40000f3d 00000023 00000000 3fffd9d0  
3fffee90:  00000000 10110101 00640104 00000060  
3fffeea0:  3ffed864 000000f0 3ffed894 3ffed858  
3fffeeb0:  3ffed881 3ffed864 3ffed86e 3ffed87b  
3fffeec0:  00000000 3ffed89a 3ffed8b6 3ffed921  
3fffeed0:  3ffed8d3 3ffed8ef 00000000 00000000  
3fffeee0:  00000000 00000000 0000002c 00000000  
3fffeef0:  3fff54dc 4025fd66 3ffeea28 3fff5234  
3fffef00:  00000000 3ffefb00 3ffeea28 3ffed840  
3fffef10:  3ffed840 00000118 00000000 0000002c  
3fffef20:  00000000 3ffed84a 40273823 3ffeea28  
3fffef30:  3ffed834 3fffdcc0 3ffeb0d8 3ffeb0d8  
3fffef40:  00000080 3ffeea28 00000000 3fffdab0  
3fffef50:  402730e7 3fffdab0 00000000 4023cb46  
3fffef60:  3ffeb0d8 40000f49 3fffdab0 40000f49  
3fffef70:  40000e19 000957bb bff00000 0000bfff  
3fffef80:  00000005 aa55aa55 000000ed 40105c91  
3fffef90:  40105c97 bff00000 0000bfff d51fc6a3  
3fffefa0:  4010000d bff00000 000957bb 401000ab  
3fffefb0:  40267c68 3fffef4c 40267c19 3ffffe58  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe  
3ffff110:  feefeffe feefeffe feefeffe feefeffe  
3ffff120:  feefeffe feefeffe feefeffe feefeffe  
3ffff130:  feefeffe feefeffe feefeffe feefeffe  
3ffff140:  feefeffe feefeffe feefeffe feefeffe  
3ffff150:  feefeffe feefeffe feefeffe feefeffe  
3ffff160:  feefeffe feefeffe feefeffe feefeffe  
[... 206 lines cut ...]
3ffffe50:  00000000 00000000 3ffe85f4 00000000  
3ffffe60:  3fffff0c 00000004 4023cc62 3fffefb0  
3ffffe70:  3fffff0c 00000000 3fff0aa0 402110fb  
3ffffe80:  3ffe8db8 00000000 00000000 fffffffe  
3ffffe90:  746e7553 6b636172 485f7265 2f30334d  
3ffffea0:  2f316863 6c656959 746f5464 00006c61  
3ffffeb0:  40222ddf 00000001 3fffffa0 00000003  
3ffffec0:  2e393135 20333231 0068576b fffffffe  
3ffffed0:  40212270 3fffc6fc 00000000 3fff0698  
3ffffee0:  3fff2c98 3fffdad0 00000000 002b002f  
3ffffef0:  00000000 00000000 000b000f 00000000  
3fffff00:  00000000 002b002f 00000000 00000000  
3fffff10:  0020002f 00000000 3fff4690 00000030  
3fffff20:  3fff0ac4 3fff2790 3fffff00 3fff06ac  
3fffff30:  3fff099c 3fff0a6c 4023cc62 3fff0a6c  
3fffff40:  3fff099c 3fff06cc 3fff069c 4020c64b  
3fffff50:  01010101 01010101 00010101 00000000  
3fffff60:  00000000 00000000 00000000 3fff0000  
3fffff70:  3fff0a6d 3fffff51 3fff0698 40100149  
3fffff80:  00000000 3fff1920 00000020 3fff4690  
3fffff90:  3fffdad0 3fff2c98 3fff0698 40222da7  
3fffffa0:  3fffdad0 00000000 3fff4664 3fff4690  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

CLAM01, 2024-10-11:
Das ist die Terminalausgabe aus dem Webinstaller.

CLAM01, 2024-10-11:
Zuvor sagte das Terminal "connection TCP lost" und hat immer Versucht sich zu connecten.
Danach kam diese Exeption.
Was die Exeption aussagt kann ich nicht sagen weil ich es nicht entschlüsseln kann.

CLAM01, 2024-10-11:
Exeption(29):
29

StoreProhibitedCause

A store referenced a page mapped with an attribute that does not permit

Region Protection or MMU

CLAM01, 2024-10-12:
Ich habe jetzt die .140 am laufen an dem Access Point von TP Link.
Versuche da auch nen Stack zu bekommen.
Wäre Klasse wenn jemand diesen dann Mal entschlüsseln kann.
Ich kann es ja nicht über die arduino ide weil ich den Sketch nicht habe.

CLAM01, 2024-10-12:
Da bekomme ich nen OOM Error.
Kann es sein, dass die 0.8.140 nicht für nen ESP8266 Wemos D1 Mini (4MB) ausgelegt ist?

CLAM01, 2024-10-25:
Dieses Thema scheint nicht mehr relevant, der TP Link scheint kein Übeltäter zu sein.
Von meiner Seite aus keine Relevanz mehr, weil eher das Thema #1766 dazu gehört und das Thema #1737

stefan123t, 2025-05-18:
@CLAM01 kann das zu oder soll da noch was analysiert werden ?

Wir haben bei OpenDTU gerade die o.a. disconnect reason der WiFi Verbindungen analysiert.
Dort gibt es scheinbar spezielle Probleme mit `2 WIFI_REASON_AUTH_EXPIRE Authentication expired`, 
die m.E. von der Arduino ESP-IDF Implementierung als **Retry** und nicht als **Reconnect** gewertet wird.

Siehe hier https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2888615142 und der Hinweis von jstammi auf einen einfach gestrickten WiFi callback handler z.B. in AquaStack: https://github.com/tbnobody/OpenDTU/issues/2400#issuecomment-2881784534

stefan123t, 2025-05-23:
@lumapu gibt AhoyDTU auch die Disconnect Reason in den Logs aus ?
M.E. hilft das die Problematik etwas genauer einzuschätzen / abzuklären.

Hier ist der Code aus der OpenDTU um die WiFi Disconnect Reason Codes auszugeben:

https://github.com/tbnobody/OpenDTU/blob/3f01b8d4b7534fd5c09e67b32d0783206e88f4c5/src/NetworkSettings.cpp#L104-L114
```c++
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
        // Reason codes can be found here: https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/esp_wifi/include/esp_wifi_types_generic.h#L79-L141
        ESP_LOGW(TAG, "WiFi disconnected: %" PRIu8 "", info.wifi_sta_disconnected.reason);
        if (_networkMode == network_mode::WiFi) {
            ESP_LOGI(TAG, "Try reconnecting");
            _lastReconnectAttempt = millis();
            WiFi.disconnect(true, false);
            WiFi.begin();
            raiseEvent(network_event::NETWORK_DISCONNECTED);
        }
        break;
```

doetnetalsof, 2025-05-31:
Gleiches Problem hier, vom ersten Tag an, als ich das aho-dtu gekauft habe, war es eine Art Russisches Roulette, ob am nächsten Tag eine WLAN-Verbindung bestehen würde oder nicht, selbst der automatische Reset macht keinen Unterschied, was immer funktioniert, ist Aus-/Einschalten, aber nach einer Weile macht man das nicht mehr.

Gibt es schon einen Hinweis, wann das nächste Update kommt, v0.8.155 ist ja schon eine Weile alt ;-)

Keep up the good work  ;-)

Gubi2023, 2025-05-31:
komisch, bei mir rennt AHOY seit Monaten anstandslos. Lediglich, wenn ich über VPN mich einloggen will und das Internet sehr schwach ist, startet sich die DTU sporadisch mit "Panic"-Meldung neu (V0.8.152)

stefan123t, 2025-06-01:
@CLAM01 Christoph geht es hier noch weiter ggf mit der Disconnect Reason im AhoyDTU code ?