lumapu/ahoy #1766: OOM Exception with esp8266 4MB Flash

CLAM01, 2024-10-12:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.140

### Github Hash

f1f4481

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Mqtt enabled
NTP Server ist Fritz
No Logger aktive for console

### Error description

Wenn die DTU gestartet wird, kommt es ziemlich schnell zu einem OOM unhandled C++ exeption und der ESP startet neu oder spannt sein AP auf.
Ich habe alle Versionen der 140 (ausgenommen Prometheus) probiert.
Alle zeigen das gleiche Problem.
Auch die .152 dev Version.

Da es auch WiFi Probleme mit alten Versionen zu geben scheint (siehe #1765 ), ist für die Ahoy DTU gerade nicht mehr zu gebrauchen.

Kein anderes Gerät auf Basis esp8266 zeigt dieses Verhalten an den vorhandenen Access Points die mit 2,4 und 5 GHz laufen.

Gubi2023, 2024-10-12:
mein ESP8266 läuft mit der -all-Version nur einige Stunden und rebootet dann neu.Nur die abgespeckten Versionen laufen länger.
(ab .140 aufwärts, ältere Versionen hab ich nicht getestet)

lille2000, 2024-10-13:
Bei mir war .90 die letze ESP8266 Version, die (halbwegs) stabil war.
Ich bin auf ESP32 umgestiegen.

knickohr, 2024-10-13:
ESP8265/6 RIP 😢

CLAM01, 2024-10-13:
Beobachte es gerade.ohne Produktion alles fein. Sobald Produktion losgeht ca. jede Stunde Neustart. Ich zeichne gerade Mal die heap Fragmentierung auf.wenn da zu wenig grosse Blöcke da sind, also Fragmentierung zu hoch, würde es das erklären.

CLAM01, 2024-10-13:
[image]
Denke es liegt eher nicht an der heap fragmentation. Diese war nur einmal hoch und hat auch erst nach ca. 1h neustart gegeben. Ich habe aber nix im Netz was in Stundentakt irgendwas macht, also nichts mit bekanntes.

fsck-block, 2024-10-13:
@knickohr 
> ESP8265/6 RIP 😢

Ganz so tot sehe ich ESP8266 noch nicht, aber vielleicht nur in "einfachen" Umgebungen.

Mein esp8266-minimal (P-en; v0.8.140; ein inverter) läuft seit 13 Tagen, vorher waren es 4 Wochen.
Free heap pendelt dabei um 19k rum.

knickohr, 2024-10-13:
Ja, wer mit einem Inverter zufrieden ist, keine MQTT und kein Display braucht, wird damit zurecht kommen.

Aber offenbar geht der Spaß jetzt schon mit 2 Invertern los.

CLAM01, 2024-10-13:
Ich habe nur zwei Inverter und mqtt. Kein Display. Lese auch die Griddaten nicht aus.

Gubi2023, 2024-10-13:
das Problem mit den Neustarts dürfte grossteils an MqTT liegen, denn ohne laufen die 8266 bei mir recht stabil. Auch die ESP32 haben ja noch Probleme mit MqTT.

CLAM01, 2024-10-13:
Ich probiere gerade noch etwas. Lasse jetzt die Version an einem anderen hybrid reapeater arbeiten.

CLAM01, 2024-10-14:
Auch wenn ich einen WR deaktiviere, um die Last zu reduzieren, schmiert die DTU am Ende komplett weg un dist nicht mehr erreichbar bis zu einem manuellen Reset per Knopfdruck auf dem ESP8266.

CLAM01, 2024-10-14:
> @knickohr
> 
> > ESP8265/6 RIP 😢
> 
> Ganz so tot sehe ich ESP8266 noch nicht, aber vielleicht nur in "einfachen" Umgebungen.
> 
> Mein esp8266-minimal (P-en; v0.8.140; ein inverter) läuft seit 13 Tagen, vorher waren es 4 Wochen. Free heap pendelt dabei um 19k rum.

Mein ESP8266 mag die Minimal auch haben, aber dann funktioniert MQTT nicht mehr. No Connection.

CLAM01, 2024-10-14:
Ich habe jetzt mal testweise bei den beiden Invertern die Abfrageintervalle von 15 auf 20 Sekunden erhöht und im Bereich MQTT den Aktualisierungsintervall auf "0" gesetzt, sodass nur überragen wird, wenn neue Daten vorliegen die auch anders sind.

Gubi2023, 2024-10-14:
Sind auch meine Einstellungen, nützt aber nur begrenzt was. Anscheinend ist der 8266 damit überfordert.

CLAM01, 2024-10-14:
Also ich habe zudem das Auslesen der Grid Daten deaktiviert.
Aktuell ist er über 2h online.
Installiert ist die 0.8.152 aktuell

CLAM01, 2024-10-14:
[image]
[image]

knickohr, 2024-10-14:
Die Grid-Daten werden nur einmal ganz zum Start der DTU ausgelesen. Denke nicht das es was ausmacht.

CLAM01, 2024-10-14:
Also bei mir lauft die DTU seither seit 4h ohne Probleme und m.E. auch performanter im Seitenaufruf.

CLAM01, 2024-10-14:
Ok, seit wieder am TP Link Repeater, Same shit wie vorher. Tja, also doch keine TP Link Repeater kompatibilität?

CLAM01, 2024-10-14:
> Sind auch meine Einstellungen, nützt aber nur begrenzt was. Anscheinend ist der 8266 damit überfordert.

Was für nen Repeater oder Box Router hast du denn wo die Connection hin läuft? Auch 2,4 und 5G aktiv?

lumapu, 2024-10-14:
Die 5GHz interessieren den ESP8266 nicht, der kann nur 2,4GHz.
Hast du denn Kanal 13 aktiv, damit sind Probleme bekannt. 
Kannst du andere ESP8266 Projekte nennen, sodass man aus denen bzgl. MqTT lernen kann? 

Das Intervall für MqTT auf 0 zu stellen ist aus meiner Sicht die einzig sinnvolle Einstellung, ich habe das Feature bis heute nicht verstanden, aber man wollte es vor langer Zeit unbedingt haben. 

Hast du weitere DTUs am laufen? Also zb. eine originale?

CLAM01, 2024-10-14:
Hi, ich habe z.b. viele Steckdosen per mqtt die nutzen openbeken. Ich habe mit denen keine Probleme an diesem Access Point.
Welche mqtt Bibliothek openbeken nutzt müsste man mal nachschauen.
https://github.com/openshwprojects/OpenBK7231T_App

Den Kanal 13 habe ich nicht aktiv auf meinen Geraten, geradeaus dem Grund weil sie nicht gut sind.

Nein ich habe nur diese eine DTU am laufen und war immer zufrieden mit dieser.
Auffallig ist halt, dass ich an Fritz egal ob Router oder Repeater kein Problem habe mehr. Deswegen überlege ich, ob der TP Link die DTU stört.

CLAM01, 2024-10-15:
Auf dem ESP8266 den ich für meine Solarnachführung gebaut habe, habe ich die folgenden Bibliotheken drin für Wifi und MQTT:
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

CLAM01, 2024-10-15:
Also ich habe heute 3 verschiedene Neustartgründe in der GUI gesehen 1) Exeption 2) System/ Softwarerestart und 3) Hardware Watchdog.

Nu bin ich absolut überfragt :-)

CLAM01, 2024-10-16:
heute ist der ESP8266 ruhiger als zuvor. Änderungen im Netzwerk? Keine...

Gubi2023, 2024-10-16:
Wundertüte DTU....Meine läuft jetzt auch seit 16 Std wieder, nachdem sie sich gestern abend x-mal kurz hintereinander neu getartet hat.🤔 (Ebenfalls mit unterschiedlichen Reboot-Reasons).

CLAM01, 2024-10-16:
Wie kann man dem esp8266 denn noch bissel mehr entlocken? Würde auch ne dev Version draufdübeln die mehr Infos raus spuckt. Oder wenn gewünscht versuchen serial Stack traces zu bekommen.

CLAM01, 2024-10-17:
Also ich habe heute nochmal nen Mitschnitt in der Arduino IDE (Serieller Monitor) gemacht.
Leider lässt sich da nix richtig rauskopieren (warum auch, braucht ja keiner....)

Aber ich habe gesehen, dass von jetzt auf gleich MQTT disconnected hat weil TCP disconnected ist.
Eine neue Verdingung mit dem WLAN aufzubauen hat die DTU nicht hinbekommen bis er anschließend ein WELCOME TO AHOY! ausgibt und wieder versucht sich zu connecten.

Packt er auch nicht und wirft dann schließlich die Exeption und bootet sich durch.

Kann es sein das sich die DTU verheddert wenn sie mal die TCP Connection verliert?

I: (#0) Payload (30)
-----
I: (#1) Radio infos: -5 -4 -5 4 -6 | t: 240, s: 226, f: 4, n: 10 | p: 255
I: (#1) TX 27 CH61, 12 ret. | 15 0B 80
I: (#1) RX  33ms | 27 CH23 | 95 01
I: (#1) RX  72ms | 27 CH03 | 95 02
I: (#1) RX 121ms | 23 CH03 | 95 83
I: (#1) Payload (42)
-----
I: com loop duration: 293ms
-----
I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect
I: (#0) Radio infos: -2 -1 -2 -2 4 | t: 233, s: 170, f: 6, n: 57 | p: 1
I: (#0) TX 27 CH75, 9 ret. | 15 0B 80
I: MQTT disconnected, reason: TCP disconnect
I: (#0) RX  40ms | 27 CH23 | 95 01
I: (#0) RX  85ms | 27 CH03 | 95 82
I: (#0) Payload (30)
-----
I: (#1) Radio infos: -5 -4 -5 4 -6 | t: 241, s: 227, f: 4, n: 10 | p: 255
I: (#1) TX 27 CH61, 9 ret. | 15 0B 80
I: (#1) RX  26ms | 27 CH23 | 95 01
I: (#1) RX  74ms | 27 CH03 | 95 02
I: (#1) RX 121ms | 23 CH03 | 95 83
I: (#1) Payload (42)
-----
I: com loop duration: 311ms
-----

-------------------
Welcome to AHOY!

point your browser to http://192.168.4.1 (Password: esp_8266)
to configure your device
-------------------
scanning APs with SSID LART0001
BSSID 0: 24 2f d0 96 a2 6c
BSSID 1: 98 9b cb 46 5f 65
BSSID 2: 20 23 51 94 b4 72
connect to network 'LART0001
try to connect to BSSID: 24 2f d0 96 a2 6c
scanning APs with SSID LART0001
BSSID 0: 24 2f d0 96 a2 6c
BSSID 1: 98 9b cb 46 5f 65
BSSID 2: 20 23 51 94 b4 72
connect to network 'LART0001
try to connect to BSSID: 24 2f d0 96 a2 6c
scanning APs with SSID LART0001
I: (#0) Radio infos: -2 -1 -2 -2 4 | t: 234, s: 171, f: 6, n: 57 | p: 1
I: (#0) TX 27 CH75, 9 ret. | 15 0B 80
I: (#0) request timeout: 300ms
I: (#0) TX 27 CH75, 9 ret. | 15 0B 80
I: (#0) RX  26ms | 27 CH40 | 95 01
I: (#0) RX  80ms | 27 CH40 | 95 82
I: (#0) Payload (30)
-----
I: (#1) Radio infos: -5 -4 -5 4 -6 | t: 242, s: 228, f: 4, n: 10 | p: 255
I: (#1) TX 27 CH61, 9 ret. | 15 0B 80
[... 175 lines cut ...]
last failed alloc caller: 0x40226185

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00093780
~ld
��n�r��n|�l�l`bbrl�nB�nl`�rl�l��I: Initializing FS ..
I:  .. done
I: Config Version: 11
I: Settings valid: true
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 76 (~ 2476 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
......

CLAM01, 2024-10-17:
Und keine 5 Minuten später, das gleiche Speil wie oben beschrieben, nur mit einem Unterschied:

Die exeption ist jetzt kein OOM sondern 29.

```
ry to connect to BSSID: 24 2f d0 96 a2 6c
scanning APs with SSID LART0001
BSSID 0: 24 2f d0 96 a2 6c
BSSID 1: 98 9b cb 46 5f 65
BSSID 2: 20 23 51 94 b4 72
connect to network 'LART0001
try to connect to BSSID: 24 2f d0 96 a2 6c

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

```

knickohr, 2024-10-17:
OOM ist out of memory. Dem Ding geht die Puste aus 😞

CLAM01, 2024-10-17:
ja, aber nicht ohne Grund.
Wenn dann müsste dem Ding immer zugleich die Puste ausgehen und dem Ding geht nur die Puste aus wenn die TCP Connection wegbricht.
Ich sehe hier eher ein Problem des Speichermanagements. Ggf. versucht die DTU trotz nicht bestehender MQTT Verbindung etc. die MQTT Nachrichten abzugeben und lässt sie in der queue und dann ist der speicher voll.

CLAM01, 2024-10-17:
Also ich für einen Teil habe nun genug Analyse betrieben.
Nu liegt der Ball nimmer bei mir.

Gubi2023, 2024-10-17:
falls du doch noch auf ESP32 wechselst, halt uns doch bitte auf dem laufenden 🙏

CLAM01, 2024-10-17:
Da warte ich nu erstmal bis hier Rückmeldungen zu meinem Problem und ggf. Lösungsmöglichkeiten kommen.
Aktuell ist es eher ein Monolog den ich hier führe. :-)

knickohr, 2024-10-17:
Changelog : 
https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md

In 0.8.150 wurde das Problem gefixed,

CLAM01, 2024-10-17:
Was wurde gefixt?
Ich habe die 152 am laufen.

lille2000, 2024-10-17:
Probier mal die 0.8.97, ob die stabiler ist? Vielleicht hilft das, den fehlerhaften Code einzugrenzen.
Seit .98 ging bei mir nichts mehr.
Die Änderungen damals waren „new network routines“.

CLAM01, 2024-10-17:
Ich hatte auch die 0.8.36 drauf.
Auch nett besser.

CLAM01, 2024-10-18:
was soll ich jetzt davon halten???
[image]

Gubi2023, 2024-10-18:
versuchs mit dem: https://fw.ahoydtu.de/tools/webflasher/

CLAM01, 2024-10-18:
https://fw.ahoydtu.de/tools/webflasher/

Hier hat es geklappt...

CLAM01, 2024-10-20:
Läuft bisher besser, aber auch hier kommt ab und zu Mal nen Wachhund vorbei.

CLAM01, 2024-10-22:
Aber da es ja nicht um den ESP32 S3 hier geht sondern um den ESP8266 meine Frage...
Wie geht es mit dem Thema nun weiter? Wird hier nochmal versucht eine Optimierung herbei zu führen?

steve-zero, 2024-10-27:
Ich frage mich auch wie es weitergehen soll. Entweder man erklärt den 8266 mit der 0.8.83 als End-Of-Life, oder man versucht wenigstens die Probleme zu lösen.
So geht die Entwicklung auf den ESP32 einfach immer weiter und es gibt keinen stabilen 8266 Build mehr.
Wobei ja auch der ESP32 mittlerweile ein paar Probleme zu haben scheint. Ich glaube dass die Entwicklung schneller geht als das Testen und Debuggen...

knickohr, 2024-10-27:
Natürlich geht die Entwicklung weiter, getrieben von euch !

Und das was mittlerweile auf einen armen ESP8266 draufgepackt wird ist schon heftig, ich sage nur Display, Webgrafik, MQTT in allen möglichen Varianten, ….

Wer selber compilieren kann sollte sich die Module bewußt und gezielt auswählen. Und ja, ich gebe Dir Recht, der Softwarestand für den 8266 sollte eingefroren werden. Wer all das nicht nutzt, kann auf diese Minimalversion ausweichen.

Beim ESP32 sieht das ein bißchen anders aus. Hier ist der limitierende Faktor der Flash und RAM. 4MB sind halt mittlerweile schon knapp geworden. Hier gibt es aber welche mit 8MB.

Und wem das nicht ausreicht, der kann einen ESP32-S3 mit Minimum 8MB nehmen. Da sind dann auch alle Funktionen mit allen Modulen und auch die zukünftige „bedarfsoptimierte Leistungsregelung“ (Zero-Export) problemlos möglich. Auch ist hier die Inverteranzahl nicht auf 4 (ESP8266) bzw. 16 (ESP32) beschränkt.

Und wer all das nicht kann oder dessen Skills nicht ausreichen, bzw. sich die ganze Hardwarefrickelei sparen möchte, dem kann ich das Fusion-Board, entwickelt in der Discord-Community, nur wärmstens ans Herz legen 😇

knickohr, 2024-10-27:
So, und jetzt @lumapu Zu mit dem Issue hier. Ich sehe keinen weiteren Diskussionsbedarf 🤓