lumapu/ahoy #1768: HSM-1000-2T not possible to configure + ESP time needs to sync at every startup

Charley2b, 2024-10-18:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.152

### Github Hash

78cbd77

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverter - add inverter: serial number 140A019A13D (note this is not the W model : HMS-1000-2T

### Error description

After entry of serial number (gets accepted), tab INPUTS only shows 1 line and RADIO shows wrong setting fields.

HMS-1000-2T has 2 solar panel connections.

stefan123t, 2024-10-18:
@Charley2b thanks for posting the Serial Number `140A019A13D`

I would expect 4 model + 3 manufacture year/week + 5 weekly serial ID = 12 characters.
Could it be that you are lacking the last or an intermediate number/character ?

`140A` may be attributed tonthe wrong model, ie a 1-on-1 model with only one DC input?

Charley2b, 2024-10-18:
Thanks Stefan, i missed a digit and is now corrected in the first message

Charley2b, 2024-10-22:
Seems that the new serial number caused this, see below link:

https://github.com/tbnobody/OpenDTU/issues/2230