lumapu/ahoy #1770: Webserver not properly responding

technics42, 2024-10-19:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.152

### Github Hash

78cbd77

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

2 inverters polling at 10s, ko MQTT

### Error description

After 2-3 days of working, this morning, the webserver/api was not responding properly any more.
Webpages were not fully populated, e.g. system-page did not show "sun" part.
Menu where to choose weppages, did only show very few entries, e.g. "live" was missing.
Api did only return "null" for all calls.
Display was showing proper values, inverters were polled, values up to date.
I had to power off/on to restart ahoy.
So this is an issue of the webserver.

Previously running on version 0.8.101 for weeks/months without any problems.

Gubi2023, 2024-10-19:
see also https://github.com/lumapu/ahoy/issues/1748#issuecomment-2365278211

MaDill, 2025-02-14:
I think, I have a similar issue. ESP8266, version 0.8.140 git f1f4481.

Suddenly the response from the Webserver is not complete. This broken webpage is displayed in normal speed. I don't use Mqtt. No display. I use the API to pull needed information.

[image]

[image]

[image]