lumapu/ahoy #1773: How Enable / Show / Test Zero-Export Config on Openfusion Board

sibelle-labs, 2024-10-20:
## Hardware
  * [ ] ESP8266
  * [X] ESP32
  * [ ] Raspberry Pi

Modelname: OpenDTU Fusion
Retailer URL: https://shop.allianceapps.io/products/allianceapps-opendtu-fusion

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [X] Voltage stabilizing motherboard

### Version / Git SHA: 
Version:0.8.152
Github Hash: 78cbd77

### Build & Flash Method:
* [X] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hi!,
Today I discovered the following link: https://docs.ahoydtu.de/de/latest/zeroExport.html
I am currently running the development version 0.8.152 on an OpenDTU Fusion.
But how do I get a look at the zero export function?
Do I have the wrong platform? or file? or what am I missing?

installed version (original filename):
241007_ahoy_0.8.152_78cbd77_opendtufusion.bin

Many thanks,

sibelle-labs, 2024-10-20:
see discord channel

Gubi2023, 2025-06-17:
Zero Export gibt es ab V0.9.xx, die ist aber noch nicht freigegeben!