lumapu/ahoy #1780: ahoy_0.8.140 display config missing

Saeniv, 2024-11-03:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.140

### Github Hash

f1f4481

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Factory setup

### Error description

Flashed the new 0.8.140 firmware and the display config was no longer available, flashed the older 0.8.83 version, and it was back and working. I have not seen any issue, but maybe most of the people are using the AhoyDTU not as standalone device.

knickohr, 2024-11-03:
Did you flash 8266 or 8266-all ?

Saeniv, 2024-11-03:
I tried esp8266-all and esp8266-prometheus, maybe I did something wrong.

Bobmorton-TX, 2024-12-07:
Hello! 
awesome project!!
I have the same issue. 
just reflashed esp8266-all

no display config is showing.

[image]