lumapu/ahoy #1782: timezone Einstellung hat keine Auswirkung

sky321, 2024-11-04:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.152

### Github Hash

78cbd77

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Only MQTT config

### Error description

Unabhängig von der timezone Einstellung läuft die Uhr 1 bzw. 2 Stunden falsch. Sync über Browser oder NTP macht kein Unterschied.
[image]
[image]

sky321, 2024-11-04:
[image]

knickohr, 2024-11-04:
Hast Du Ahoy nach dieser Änderung auch rebootet ?

Gubi2023, 2024-11-04:
ich kann obiges Verhalten so bestätigen. Zusatz: nach reboot geht die zeit im **zuvor geöffneten** Webserial um 1 Std. nach!

sky321, 2024-11-04:
Ja klar, nach jeder Änderung immer mit reboot.

sky321, 2024-11-04:
Interessanter Weise sitze ich gerade am Ipad und da stimmt die Zeit. ABER die timezone Einstellung hat auch da keinerlei Auswirkungen. Vorher habe ich am Android Tab gearbeitet.

Gubi2023, 2024-11-04:
habe soeben festgestellt, dass sich die Systemzeit-Anzeige nur mit refresh aktualisiert  
[image]

sky321, 2024-11-04:
Refresh macht bei mir kein Unterschied.

sky321, 2024-11-08:
Wie kann es den sein das die aktuelle Zeit von Ahoy unterschiedlich dargestellt wird abhängig davon welchen client und/oder Browser ich benutze?
Android 14 Mull Browser
[image]