lumapu/ahoy #1783: Kondensator Wert / Capacitor Value

CK1-CK, 2024-11-04:
Das ist kein Issue, sondern eine Frage.

Warum wurde an diese Stelle https://github.com/lumapu/ahoy/blob/2d5d62eb0e2b35c743db2b75079fb039c0f9c191/manual/Getting_Started.md?plain=1#L73 ein 100uF Kondensator ausgewählt? 
Gibt es dazu einen Hintergrund? 
Kann man auch einen 1000uF Kondensator nehmen?

Klar der Kondensator muss anfangs vom Spannungsregler geladen werden. 

So wie ich es verstehe, soll der Kondensator als Spannungsstabilität dienen. 

Vielen Dank schon im Voraus.
CK

########English########

This is not an issue, but a question.

Why was a 100uF capacitor selected for this position? 

https://github.com/lumapu/ahoy/blob/2d5d62eb0e2b35c743db2b75079fb039c0f9c191/manual/Getting_Started.md?plain=1#L73

Is there a background to this? Can a 1000uF capacitor also be used? Of course the capacitor has to be charged by the voltage regulator at the beginning. 

As I understand it, the capacitor should serve as a voltage stabilizer. 

Many thanks in advance.
CK

stefan123t, 2024-11-05:
Siehe hier https://github.com/lumapu/ahoy/issues/1064, hier https://github.com/lumapu/ahoy/issues/1217 und hier https://github.com/tbnobody/OpenDTU/discussions/1047
Und natürlich in all unserer Dokumentation.

kiu77, 2024-11-05:
Hintergrund:
Je größer ein solcher "Kondensator" (der faktisch ein Elko = Elektrolytkondensator ist), desto größer ist nicht nur seine Kapazität, sondern auch seine Induktivität. Letzteres reduziert quasi das Puffern sehr schneller Änderungen einer anliegenden Spannung. Folglich sind 100µF hier ein guter Kompromiss zwischen den diversen, sich teilweise widersprechenden Eigenschaften - und ein 1000µF-Teil muss nicht besser filtern.

CK1-CK, 2024-11-05:
> Hintergrund: Je größer ein solcher "Kondensator" (der faktisch ein Elko = Elektrolytkondensator ist), desto größer ist nicht nur seine Kapazität, sondern auch seine Induktivität. Letzteres reduziert [...]

Danke für die Antworten. 

Ich bin jetzt kein Elektro-Spezialist aber ich dachte eigentlich, dass bei DC Gleichstrom keine Induktivität zum tragen kommt. 
Eigentlich sollte doch bei DC ein nahezu unendlicher Kurzschlussstrom (begrenzt durch den Serienwiderstand) zur Verfügung stehen.

Ich werde mal die SPI Verkabelung verdrillen und schauen ob das hilft. Falls das nicht hilft, werde ich auf einen 100uF Kondensator wechseln. 
Ich habe nämlich das Problem, dass mein ESP8266 regelmäßig komplett aussteigt. Dann hilft nur noch Spannung entfernen und neu starten.

Ersatzschaltbild Kondensator:
https://www.elektroniknet.de/e-mechanik-passive/passive/wenn-ein-kondensator-zur-spule-wird.201687.html

Video mit RF24 Lösungen
https://www.youtube.com/watch?v=Z7_Cy66Vnrc

stefan123t, 2024-11-06:
@CK1-CK es ist halt auch nur Gleichstrom so lange die Belastung gleich bleibt. Da wir aber, vor allem beim Senden mit den NRF24L01+ Modellen mit PA+LNA, einen kurzfristig hohen Bedarf haben, der evtl. auch über zu dünne Dupont Wires nicht schnell genug gedeckt werden kann, schwankt dieser "Gleichstrom"-Bedarf eben. Und das kann dann auch wieder zu den o.a. Effekten von @kiu77 führen.

Aber wenn Dein ESP8266 komplett aussteigt, dann vermute ich eher einen Programm / Memory-Fehler. 
Gerade für den ESP8266 wurden in letzter Zeit einige Issues aufgemacht.

Verwendest Du die -all oder die -minimal Variante des ESP8266 Images ?

CK1-CK, 2024-11-06:
Leider kann ich den Fehler nicht reproduzieren. 

Ich verwende das Standard DE-Image (erste Zeile) weil ich die Werte mittels mqTT weitergeben muss. 

|       FW Tag/Feature       | MqTT  | Display | History/Verlauf | Prometheus |
|--------------------|-------|---------|-----------------|------------|
| **esp8266**            | X     | -       | -               | -          |
| esp8266-all        | X     | X       | X               | -          |
| esp8266-minimal    | -     | -       | -               | -          |
| esp8266-prometheus | X     | X       | X               | X          |

Eventuell spielt auch die Memory Fragmentierung eine Rolle.

Leider haben wirklich viele Leute Probleme mit dem ESP8266 und NRF24.

Im Moment ist das System so unzuverlässig, dass man es leider nicht produktiv nutzen kann.

Jumperm, 2024-11-17:
> …
> Ich verwende das Standard DE-Image (erste Zeile) weil ich die Werte mittels mqTT weitergeben muss.
> 
> Eventuell spielt auch die Memory Fragmentierung eine Rolle.
> …
> Leider haben wirklich viele Leute Probleme mit dem ESP8266 und NRF24.
> 
> Im Moment ist das System so unzuverlässig, dass man es leider nicht produktiv nutzen kann.

Ich nutze die minimal Version und habe eigentlich (WLAN Verbingungsprobleme, wegen MESH) keine Probleme.

Wenn mein D1mini einmal mit dem WLAN verbunden ist, wird er per REST API alle 3 Sek abgefragt. Er selbst sammelt die Infos des HM alle 3 Sek. ein.

```
Config Partition (24kB of 1000kB)	
Firmware Partition (558kB of 2512kB)	
Heap (21kB of 24kB)	
Heap max free block	21kB (Fragmentation: 3)
```

CK1-CK, 2025-01-01:
Moin, 

Ich hatte jetzt endlich Mal Zeit ein bisschen zu basteln. 

Der Fehler ist, nachdem ich die MISO und MOSI Leitungen verdrillt habe, nicht mehr aufgetreten. Ich werde das jetzt Mal über einen längeren Zeitraum beobachten.

Immer wieder toll, was AC/Hochfrequenz für Auswirkungen haben kann. 

Wahrscheinlich waren die dünnen parallelen Leitungen (ohne Verdrillung) selbst ein Kondensator.

Vielen Dank für eure Unterstützung! 
Immer wieder toll, was die Community alles erreichen kann.

closed 2025-01-01 by CK1-CK