lumapu/ahoy #1786: Inverter disabled

themille, 2024-11-14:
## Hardware
  * [ ] ESP8266
  * [ x] ESP32
  * [ ] Raspberry Pi

Warum ist der Wechselrichter deaktiviert, aktivieren kann ich ihn nicht auch wenn ich den Haken setze und speicher.
[image]
[image]

Gubi2023, 2024-11-14:
bitte mehr Infos: welche DTU, welche Version? ggf. Screenshot von "System" senden

Gubi2023, 2024-11-14:
Interrupt unnown:  vermutlich pinmapping falsch

themille, 2024-11-14:
[image]

knickohr, 2024-11-14:
Also ich sehe hier schon mal mindestens 2 Probleme : 
- der Interrupt vom CMT arbeitet nicht
- warum wird der Flash nich5 angezeigt ?

Zeig uns auch mal die Settings mit den LEDs 😉

[image]

themille, 2024-11-14:
[image]

knickohr, 2024-11-14:
Jetzt komm schon, etwas mehr Info.

Welche Hardware, wie verkabelt, gekauft oder selber zusammen gestrickt, welche Module …

Bilder vom Aufbau wären auch nicht schlecht.

Meine Glaskugel arbeitet heute noch nicht, es ist kalt 😅

Gubi2023, 2024-11-14:
woher hast du deine Pinbelegung? bei mir sieht sie ganz anders aus...

themille, 2024-11-14:
Man ich bin doch totaler Anfänger.
Ich habe ein ESP32WROOM32 und ein CMT2300A von blinkyparts gekauft.
Habe diese beiden so verkabelt:
[image]
Mehr habe ich nicht gemacht.
Pinmapping dann so eingestellt:
[image]
Ist die Pinbelegung von blinkyparts

themille, 2024-11-14:
ok, das Problem dass der WEchselrichter nicht genommen wird, liegt defintiv nicht an der Pinmapping.
Ich habe zwei WR vom TYP HM600 eingegeben und sie werden deaktiviert. Nur der HMS800 wird direkt deaktiviert nach Eingabe seiner Serialnummer (die sind ja mit Buchstaben)

knickohr, 2024-11-14:
Der HMS ist aber ohne ein W in der Beteichnung ?

knickohr, 2024-11-14:
Ich hau Dich gleich !
Das ist ein WiFi-WR, die funktionieren weder mit Ahoy noch mit OpenDTU 😞

themille, 2024-11-14:
alter sorry, das habe ich nirgends gefunden, tut mir leid

knickohr, 2024-11-14:
Auf der Gihub Seite von Ahoy.

Und in der Doku ist es auch drin : 

[image]

knickohr, 2024-11-14:
> alter sorry, das habe ich nirgends gefunden, tut mir leid

Alles fein, wird leider gerne überlesen 😇

Gubi2023, 2024-11-14:
@knickohr : Vorschlag vllcht noch auf discord.com  #faq-doku  verlinken?  Ich hab da leider keinen Zugriff...

rmayergfx, 2024-11-20:
Also ehrlich, das steht dick und Fett auf der Homepage:
**November 2023: HMS-xxxxW Serie

Leider kann Ahoy die Wechselrichter der HMS-xxxxW Serie (noch) nicht auslesen. Diese Wechselrichter haben eine integrierte DTU und kein Funkmodul mehr wie die drei vorherigen Generationen.**

closed 2024-11-20 by lumapu

stefan123t, 2024-11-20:
> Meine Glaskugel arbeitet heute noch nicht, es ist kalt 😅

[image]