lumapu/ahoy #1787: Störung Nachtkommunikation mit HMS Invertern

tester277, 2024-11-16:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.152

### Github Hash

-

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Habe ein Fusion Komplett Gerät mit ESP32S3, daran sind insgesamt 4 WR(ausschließlich HMS) angeschlossen.

### Error description

Beim Betrieb in der Nacht gibt es nach einigen Tagen Kommunikationsaussetzer. Im  webserial sind hohe com loop Zeiten >1000ms es werden teilweise keine Daten angezeigt und die Lossraten gehen hoch.
z.B.
TX count | 44230 |  
-- | -- | --
23419 | 52.95 %
0 | 0 %
20811 | 47.05 %
55461 |  
17663 |  
lost 445 of 554 | 80.32 %
lost 65448 of 119 | 54998.32 %
Tagsüber gibt es solche Probleme nicht.
Wenn so etwas passiert, lässt sich das auch nicht z. B. über ein Neustart beheben. Bei Tageslicht ist dann das Problem weg.
Einen Auslöser (außer Nachtbetrieb) habe ich nicht entdeckt. Was kann ich zur Untersuchung beitragen?

knickohr, 2024-11-17:
Sind alle Inverter betroffen ?  Oder nur einzelne, was sind das für welche ?

knickohr, 2024-11-17:
Das weiß ich schon.

Sind die Inverter die Aussetzer haben mehrkanalige ?

tester277, 2024-11-17:
Beteiligt sind ein HMS 500 da sind 2 parallel geschaltete 24V Akkus dran ( also ein Channel) und ein HMS 1800 Channel/Input 1: PV Modul; Channel/Input 2 und 3 Akkusystem Growatt Noah; Channel/Input 4 nicht belegt.

knickohr, 2024-11-17:
Bei dem 500er wäre es vielleicht sinnvoll aus den 24V-Batterien ein 48V-System zu machen.

Beim 1800er (ich denke es ist ein 4T). Mal den Input 4 belegen, mansche haben nur auf einem Input die Versorgung der Elektronik.

tester277, 2024-11-17:
Zu Antwort 1) 48V System ist sicherlich eine gute Lösung, ich habe aber nur ein 24V Ladegerät. Ich glaube nicht, dass das der Auslöser für oben beschriebenenes Problem ist.
Zu Antwort 2) Der HMS hat 4Mppt. Tagsüber kommt der Inverter mit der Versorgung über Input 1 ohne Loss Rate klar.  
[image]
Im Bild zu sehen die Watt Werte der beiden WR nach Sonnenuntergang (heute bei mir 17.29). Kurz nach 19 Uhr wieder normale Kommunikation. Das geht in der Nacht so weiter.

knickohr, 2024-11-17:
Das ist korrekt, aber nur einer versorgt die Elektronik im WR. Frag mal Dahlienhaus, der hatte das gleiche Problem das sein HMS, ich glaube es war ein 1800-6T Nachts immer aus ging.

tester277, 2024-11-20:
Vielen Dank @knickohr für Deine unermüdliche Unterstützung hier. Wahrscheinlich werden die Werte nur korrekt erfasst und übermittelt solange alle WR bei denen die Einstellung „Pause communication during night“ deaktiviert ist auch tatsächlich online sind. Sonst kommt es zu solchen Aussetzern wie oben beschrieben, weil immer versucht wird, WR die gerade offline sind mit abzufragen. Ähnliches Thema: #1785.

tester277, 2024-11-20:
Wahrscheinlich werden die Werte nur korrekt erfasst und übermittelt solange alle WR bei denen die Einstellung „Pause communication during night“ deaktiviert ist auch tatsächlich online sind. Sonst kommt es zu solchen Aussetzern wie oben beschrieben, weil immer versucht wird, WR die gerade offline sind mit abzufragen. Ähnliches Thema: https://github.com/lumapu/ahoy/issues/1785.

closed 2024-11-20 by tester277