lumapu/ahoy #1788: e-paper probleme

sommer05, 2024-11-19:
### Platform

ESP32

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.140

### Github Hash

f1f4481

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Wechselrichter Intervall = 10 sec

### Error description

Ahoy an alle ahoy'ler 
ich habe einen ESP32 WROOM32 erfolgreich mit ver 0.8.140 geladen und habe ein 1.54" e-paper Module V2 Rev 2.1 von Waveshare darauf installiert und eingerichtet.
das ganze wurde über ein board von 2Mars zusammengebaut
[image]

das e-paper wurde wie folgt konfiguriert
[image]
alles startet soweit gut , Ahoy Logo kommt zuerst kurz dunkel, 
[image]
dann hell
[image]
nach ein paar sekunden dann die verbindung zur DTU mit starken Kontrast und gut lesbarer schrift (dzt offline weil abend - macht aber tagsüber den gleichen effekt)
[image]
nach ziemlich genau 14 sec ein sichtbarer refresh und die schrift wird fast unleserlich, kontrastlos.
[image]
unter tags dann auch noch teilweise dunkle teile der schrift ersichtlich
[image]
habe schon mit sämtlichen alten version, auf engl oder deutsch rauf und runtergeladen aber überall der selbe effekt.
ich finde auch in älteren einträgen nicht wirklich einen eintrag dazu.
ist das ein fehler ? habe nur ch diesen effekt ?
vielleicht kann mir da ja jemand weiterhelfen....
achja und ich habe ein starkes 2,5A netzteil verwendet
lg aus wien

knickohr, 2024-11-20:
Ist ein runder grauer Aufkleber mit V2 drauf und was steht auf dem brauchen Flachbandkabel. Leider wechseln sie öfters die eigentlichen Displays (Glas), so das bisher wohl nur ein einziger Typ funktioniert 😞

[image]

akrotky, 2024-11-20:
hallo
v2 pickerl ist drauf und am flachbandkabel steht 
FPC B001 
20.05.21
was steht bei den versionen die funtionieren oder wo hast du deine her ?
danke im voraus

knickohr, 2024-11-21:
[image]

Nicht erschrecken lassen von dem grauen Display, das Modul wa4 seit Wochen nicht mehr angeschlossen 😅

knickohr, 2024-11-21:
Direkt vom Hersteller :  waveshare

akrotky, 2024-11-21:
ok danke, werde mal eines ordern und schaun obs damit läuft.
wobei mir nicht klar ist warum es am anfang dunkel ist und erst nach den ertsen refresh hell wird- eine treibergeschichte kann das nicht sein ? oder ?
lg

knickohr, 2024-11-21:
Doch, es ist vermutlich eine Treibersache, je nach Library die verwendet wird 😞

sommer05, 2024-11-21:
ok dann meine letzte frage,ich verwende bisher die fertig compilierte .bin von ahoy. wenn ich die „richtige“ library von der wavesh. page finde könnte ich die ja selbst austauschen und selbst compilieren.welche ist denn diese in diesen repositurium?hoffe habe mich richtig ausgedrücktlg armin

knickohr, 2024-11-21:
Wenn Du der Sprache mächtig bist 7nd auch die Aufrufe für die Library ändern kannst, dann gerne.

Ich fürchte aber damit ist es nicht getan, selbst die Funktionsaufrufe werden nicht miteinander kompatibel sein, geschweige denn die gesetzten Übergabeparameter.

Sollte ich mich aber irren, dann gerne, vielleicht kommt wieder etwas Leben in die Sache ePaper und vielleicht funktionieren dann auch sndere/größere Displays 😉

Loetnase, 2024-11-25:
Falls du es einfach haben willst kaufe hier ein ePaper 
https://shop.semy3d.de/Waveshare-154inch-e-Paper-Module
Das ist unser Partnershop welcher auch komplette DTUs aufbaut und vertreibt, 
der hat auf jeden Fall die richtigen Displays. 
Aber wenn du dich hier um die ePaper bzw. auch andere Displays kümmern könntest, wäre das eine tolle Sache.
Du kannst dazu auch gerne einmal im Discord dazu vorbeischauen.

sommer05, 2024-11-26:
mal sehen ob meine bescheidenen programmierkentnisse dafür reichen, werde mich mal daran versuchen - bischen revers engineering ist dafür wohl notwendig um alle display module zu verstehen....
lg

Loetnase, 2024-11-26:
Vielleicht kann dir @knickohr den Kontakt des ePaper Erstellers dann weiter geben, 
sobald du tieferes Wissen dazu brauchst. 
Revers engeniering ist da mit Sicherheit einiges notwedig, aber vieleicht kann dazu auch etwas Hilfestellung gegeben werden.
Viel Erfolg

silverseraphpirrs799, 2024-12-18:
Hallo sommer05, sobald du hier eine Lösung hast, lass mich bitte teilhaben, ich habe genau die gleiche Problematik
Grüße

silverseraphpirrs799, 2024-12-18:
> Was steht bei dir auf dem Flachbandkabel vom Display? FCP-8101 oder was anderes, siehe Bild oben von Knickohr.

[image]

Loetnase, 2024-12-18:
Da kannst du nur versuchen es umzutauschen gegen eines mit der richtigen Flachbandkabelbeschriftung.
Oder 3 Post weiter oben von mir mit dem Link ein anderes kaufen, seine sollten funktionieren, er vertreibt die kompletten DTUs da merkt er wenn sie falsch sind.
Lass uns deine Lösung wissen.

sommer05, 2024-12-21:
Hallo an alle Mitleidenden,
1) nachdem ich bisher noch nicht die Zeit hatte und warscheinlich auch das tiefe Wissen  bin ich den leichteren Weg gegangen und habe 
2) 2 E-Paper direkt bei waveshare bestellt und
3) siehe da die funktionieren so wie sie sollen.
jetzt kommts aber 
die haben 
4) genau die gleiche Beschriftung auch am gelben Flachbandkabel wie die beiden "nichtfunktionierenden".
meine Frage an waveshare ob es da firmware unterschiede gibt und ob man die evtl selbst nachschiessen kann, habe ich leider eien Absage bekommen.
dürfte also wirklich an
5) unterschiedlicher HW liegen die verbaut wurde.

nicht sehr befriediegende aber auch ein weg all die defekten an die händler als offensichtliche ausschussware (billige ? ) zurückzu senden.

knickohr, 2024-12-21:
Auch das Datum auf den Kabel ?

akrotky, 2024-12-21:
Ja , alle 20.05.21

andred, 2025-05-03:
FYI, I've had similar problems as described in the first few post, but was able to finally resolve them. For me, I've always had an ePaper since v0.6.0 of ahoy, and it always worked fine until one of the updates (don't remember which version exactly).

In my case, the inherited ePaper config and Ahoy's newly added CMT radio config share some of the pins. Despite CMT radio being unused and disabled in my case, something still seems to interfere. By changing the pinout of the CMT radio in the config the ePaper started operating normally again. Note that I don't use the CMT radio and it's disabled, but still seems to interfere:

default config:

CMT2300A:
| signal | gpio |
| ------ | ---- |
| SCLK | GPIO12 |
| SDIO | GPIO14 |
| CSB | GPIO27 |
| FCSB | GPIO26 |
| GPIO3 | GPIO34 |

ePaper:
| signal | gpio |
| ------ | ---- |
| CLOCK | GPIO13 |
| DATA | GPIO14 |
| CS | GPIO15 |
| DC | GPIO27 |
| RESET | GPIO26 |
| BUSY | GPIO25 |

Note how GPIO14, GPIO27, and GPIO26 are shared.

After changing CMT to:
| signal | gpio |
| ------ | ---- |
| SCLK | GPIO12 |
| SDIO | GPIO17 |
| CSB | GPIO21 |
| FCSB | GPIO22 |
| GPIO3 | GPIO34 |

the ePaper is working fine now.

Again, I'm not using CMT and it's disabled in the config, but sharing pins still interferes

andred, 2025-05-03:
FYI, I've had similar problems as described in the first few post, but was able to finally resolve them using v0.8.83. For me, I've always had an ePaper since v0.6.0 of ahoy, and it always worked fine until one of the updates (don't remember which version exactly).

In my case, the inherited ePaper config and Ahoy's added CMT radio config share some of the pins. Despite CMT radio being unused and disabled in my case, something still seems to interfere. By changing the pinout of the CMT radio in the config the ePaper started operating normally again. Note that I don't use the CMT radio and it's disabled, but still seems to interfere.

default config:

CMT2300A:
| signal | gpio |
| ------ | ---- |
| SCLK | GPIO12 |
| SDIO | GPIO14 |
| CSB | GPIO27 |
| FCSB | GPIO26 |
| GPIO3 | GPIO34 |

ePaper:
| signal | gpio |
| ------ | ---- |
| CLOCK | GPIO13 |
| DATA | GPIO14 |
| CS | GPIO15 |
| DC | GPIO27 |
| RESET | GPIO26 |
| BUSY | GPIO25 |

Note how GPIO14, GPIO27, and GPIO26 are shared.

After changing CMT to:
| signal | gpio |
| ------ | ---- |
| SCLK | GPIO12 |
| SDIO | GPIO17 |
| CSB | GPIO21 |
| FCSB | GPIO22 |
| GPIO3 | GPIO34 |

the ePaper is working fine now.

Again, I'm not using CMT and it's disabled in the config, but sharing pins still interferes. Above solution works for v0.8.83, but doesn't help with 0.8.155. I haven't tried any versions in between yet

Loetnase, 2025-05-03:
@andred look to my Setup, i use the ESP32 with NRF and ePaper on 0.8.148, 
i use no CMT and no Ethernet but you see the standart setup there.
Your ePaper Clock and Data are turned, to the standart like mine.
I hope that helps you 
[image]

andred, 2025-05-04:
@Loetnase , thanks for sharing. This morning, my e-paper was weird again, the display also wasn't being reset properly during boot of ahoy-dtu. The only way to restore it was to disable it in the config, and re-eneable it (with same pin config).

So then I went and changed the ePaper's 'busy' pin to match yours. And surprise - things seem to working fine, even after undoing my earlier modification of CMT/Ethernet pins (i.e restoring them to default).

IOW, for me, changing ePaper's busy pin from gpio25 to gpio12 seems to have finally resolved the problems. Not sure why that should make a difference, but it appears to me that it does.

Loetnase, 2025-05-04:
@andred, I think GPIO 12, 13 +14 belong together to the **HSPI** Port, GPIO 25 is only an other GPIO 
That's like the GPIO 18, 19 +23 belong together to the **VSPI** Port whitch is used for the NRF24

[image]

andred, 2025-05-04:
Thanks @Loetnase, I didn't know that. Looks like my problem was different to the original author's after all. Sorry for that.

Loetnase, 2025-05-05:
It is possible to use all the CMT, NRF24 and ePaper, but then no ethernet. It should work with the pinning in my first picture on the ESP32. 
The ePaper and CMT share some GPIO, that is okay.
Sorry i dont have a schematik of it for you.