lumapu/ahoy #1790: No connection to HMS-1600-4T

mortiii, 2024-11-21:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.152

### Github Hash

1

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

i tried many frequencies combined with some db values (5 db steps)

### Error description

ESP 32 + CMT2300a (set from Blinky Parts) HMS-1600-4T Inverter without WIFI Module implemented. The CMT2300a is connected and all seems to be fine. i tried firmware 0.8.83 (no letters allowed in Serial Number) / 0.8.140 / 0.8.152 and the connection to the inverter was never established. i checked the serial number many times and tried many combinations of frequency and send power. The distance between inverter and esp32 is lower than 10 meters and there is no wall between.

stefan123t, 2024-11-21:
Does the CMT2300A Interrupt show it is working in the UI System Info ?
Did you connect the GPIO3 of the Blinky Parts breakout board ?
I am not 100% sure though I believe this is/was not necessary for OpenDTU but it is a requirement for AhoyDTU.
Can you provide a picture of your hardware ?

mortiii, 2024-11-29:
No one has an idea what it could be?

mortiii, 2024-12-14:
Ok i tried openDTU and connection to my HMS-1600-4t works.

TheSCC1992, 2024-12-26:
@mortiii see [https://github.com/lumapu/ahoy/issues/1749] is the same problem, no solution. I changed to opendtu and there are no problems.