lumapu/ahoy #1791: !!! Deye deaktiviert Solar Manager in USA UK und Pakistan !!! Achtung !!!

rmayergfx, 2024-11-27:
https://www.borncity.com/blog/2024/11/27/deye-deaktiviert-solar-manager-in-usa-uk-und-pakistan/

Gut wenn man gleich im Vorfeld die WR nicht in der Cloud betreibt und den Zugang zum Internet gesperrt hat.

kiu77, 2024-11-29:
Uiuiui!
Das kann ja Eiter werden!
Wenn man bedenkt, was da im Falle einer Krise (USA vs. CN) auf D zukommen kann, kann einem übel werden.

primetimber, 2024-12-01:
Hier gibt es ein Update dazu https://solarboi.com/2024/11/30/deye-addresses-bricked-inverter-controversy/
Klingt irgendwie *kompliziert*.

rmayergfx, 2024-12-01:
Gibt auch was bei ct zu diesem Thema:
[https://www.heise.de/news/Photovoltaik-Deaktivierte-Deye-und-Sol-Ark-Wechselrichter-in-den-USA-10183706.html](url)

kiu77, 2024-12-03:
@rmayergfx 
Da freue ich mich doch schon auf entsprechende Meldungen bei Autos, die definitiv generell OTA-Fähig sind seit Neuestem (wie ich bei meinem BJ 24 feststellen musste - zum Glück nicht chinesisch - aber beinahe wäre es eines geworden). Ich mag mir garnicht ausmalen, was da in Folge eines Handelskriegs machbar ist. Dann liegt die ganze Hoffnung bei Hackern ;-)