lumapu/ahoy #1792: ESP32 (38Pin) CMT + NRF - NRF module suddenly disconnected

RedJohn14, 2024-11-28:
Hello everyone,

I have setup my AhoyDTU like this. With CMT Module and NRF Module. 
https://ahoydtu.de/img/fritzing/esp32-38-nrf_cmt_sch.png

I bought and connected all the necessary components as described on the website. I can also read out both inverters for a while and everything works perfectly.

This is the “actual status”. After an undefined period of time, the NRF24 module is suddenly disconnected and one of the inverters can no longer be queried.
The CMT2300A module has no problems and is always connected.

Does anyone have any idea where the problem could be coming from? What can be done to prevent it? How can I narrow down the sources of the error?

It would be great if someone had an idea or a tip. Maybe someone has had the same problem.

[image]

Gubi2023, 2024-11-28:
bleibt die Verbindung unterbrochen, oder kommt sie automatisch wieder? Ich würde auf ein Problem mit dem NRF-Modul tippen. Entweder Schrott-Modul oder Wackelkontakt? (Modul verdrahtet oder verlötet?) Oder es verliert wirklich die Verbindung zum WR wg. Bewölkung o.ä.

lille2000, 2024-12-17:
Vielleicht habe ich das selbe Problem (ESP32, Nrf24; 3 Inverter, SW 0.8.152):
Heute (17.12.) ist mir aufgefallen, dass die letzte Übertragung von den drei invertern am 15.12 um Sonnenuntergang passiert ist. Der Ahoy System läuft seit 59d12h, der Fehler ist nach 57d aufgetreten.
Ich vermute einen Zählerüberlauf des ms Timer:
2^32=4.294.967.296 
Geht man von einem ms Timer aus, dann wird er nach 50 Tagen überlaufen ( sofern er genau kalibriert ist)
In diesem Fall war es:  
57 Tage * 24 * 3600 * 1000 ms = 4.924.800.000  (mindest mal die selbe Grössenlrdnung)

da liegt es Nahe, beim Überlauf zu suchen.

( leider bin ich zu „solar-Zeiten“ nicht zuhause, es wird schwer, Logs aufzuzeichnen.)
Ich meine, vor 59d hatte ich das Problem schonmal und hab neu gestartet und dann ging’s wieder.