lumapu/ahoy #1794: Receiving 0 via MQTT for Yield Total about twice a week

AAPohl, 2024-12-07:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Extertal MQTT Broker with password connected. That runs in Hause on RPi.

### Error description

Basically, everything works as expected. I get several data via mqtt on a regular base. However, yield Total, that should increase over time sends a zero from time to time. Typically at night. Happens around once or time a week. 

If i can add more information let me know, but it is really hard to fix at client side.

stefan123t, 2024-12-07:
This is received during dusk / sunset as the inverter may already be powered down and wakes up again. So for the inverter it is a new dawn and the Yield will be 0 again.

Maybe we ahould consider to reset the value at midnight only / always. For the rest of the day we should calculate with deltas. And if the delta is zero / negative just re-init the inverters yield counter from our memory plus the tiny yield reported as it is either rolling over or having restarted.

AAPohl, 2024-12-08:
I like the idea. I use the mqtt input for evcc and there it is joined with other inverters. Thus, it is not easy to identify the errors afterwards. 

Is it even more simple to skip all exact zeros per inverter? As far as i understand, the inverter only wakes up on pv energy, thus a yield Total of exactly zero is only an academic case?

AAPohl, 2024-12-14:
I did an update to the last develop Version and it got worse. Much more zeros on way