lumapu/ahoy #1795: Ahoy-DTU spannt Netzwerk mit 5 GHz auf!

Gubi2023, 2024-12-18:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.152

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

0.8.152

### Error description

seit einiger Zeit habe ich das Phänomen, dass meine DTU (von AllianceApps) nicht mehr erreichbar ist. Nach Spannungswiederkehr spannt die DTU ihr Ahoy-DTU-Netzwerk auf und in der Fritzbox wird mir folgendes angezeigt:
[image]
Eine Zuweisung zu meiner Fritzbox geht dann auch nicht.

knickohr, 2024-12-18:
Das wäre geil !  Kannst Du mal die genaue Bezeichnung des ESPs, am Besten Foto davon machen. Ich weiß das neuere ESPs auch 5GHz können, kann gut sein das die letzte Charge des S3 es nun auch können 😎

Gubi2023, 2024-12-19:
hallo @knickohr , ich glaub ich muss dich da enttäuschen: ich hab die DTU schon im März gekauft, dürfte eigentlich kein S3 sein.   
[image]

Wieso der trotzdem 5GHz macht, weiss ich nicht

knickohr, 2024-12-19:
Ach, das ist gar kein Fusion-Board 😅

Nee, der kann definitiv KEIN 5GHz !

Loetnase, 2024-12-21:
Frag mal bei AZ-Delivery nach, 
vielleicht kennen die Details dazu 😁

Gubi2023, 2024-12-25:
Falscher Alarm! DTU wars doch  nicht.

closed 2024-12-25 by Gubi2023