lumapu/ahoy #1800: Seriennummer mit Großbuchstaben werden nicht richtig gespeichert

hajulied, 2025-01-28:
## Hardware
  * [ ] ESP8266
  * [ x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ x] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _0.8_.140__
Github Hash: _______

### Build & Flash Method:
* [ x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

hajulied, 2025-01-28:
habe einen HMS 1600 4T bekommen. Nachdem ich auf 0.8.140 upgedatet habe, konnte ich die Seriennummer mit enthaltenen Großbuchstaben eingeben. Es scheint aber, dass Ahoy die Buchstaben als Kleinbuchstaben speichert. 
Beispiel: Eingabe und Anzeige 123B456CDE
nach dem Speichern:                123b456cde
Vermutlich dadurch verbindet ahoy sich nicht mit dem WR.

Gubi2023, 2025-01-28:
bist du sicher, dass es kein HMS **1600W** 4T, also mit integriertem WLAN ist? Den kann AHOY nicht auslesen

hajulied, 2025-01-29:
Moin, 
es ist definitiv ein HMS 1600 4T (sorry ich hatte urspr. 1800 geschrieben).

[HMS 1600 Screenshot.pdf](https://github.com/user-attachments/files/18587872/HMS.1600.Screenshot.pdf)

hajulied, 2025-01-31:
ich habe die Einstellungen als JSON heruntergeladen. Dort sind auch die Großbuchstaben als kleine eingetragen. Nach Änderung in Großbuchstaben wieder die Datei hochgeladen - aber leider kein Erfolg. Sie werden wieder klein gespeichert. 
Was kann ich denn sonst noch machen?

Gubi2023, 2025-01-31:
wie im Discord erwähnt: Verdrahtung (kalte Lötstelle) oder CMT fehlerhaft.

hajulied, 2025-01-31:
Verdrahtung ist ok, habe alle Anschlüsse durchgemessen ("durchgeklingelt😁'. CMT defekt - kann natürlich sein, aber das löst nicht die Frage, warum ahoy die groß eingegebenen Buchstaben klein speichert. Auch nach json Download, Korrektur, Reset ohne WLAN und upload der geänderten json Datei sind die Buchstaben wieder klein? Dann kann doch keine Verbindung hergestellt werden.

Gubi2023, 2025-01-31:
wenn es  bei anderen funktioniert, spielt die Gross/Kleinschreibung wohl keine Rolle würd ich sagen.