lumapu/ahoy #1801: an der Temperaturanzeige wird Datum und Uhrzeit nicht mehr aktualisiert

Gubi2023, 2025-02-01:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.152

### Github Hash

???

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Web-UI

### Debug Serial Log output

```shell

```

### Error description

Die Uhrzeit und das Datum werden bei der Temp.anzeige nicht mehr aktualisiert Siehe Screenshot

[image]

Gubi2023, 2025-02-01:
merkwürdig: scheint sich nur bei minus-Temperaturen nicht zu  aktualisieren, bei plus-Graden geht es🤔

Dirk-Ahoy, 2025-02-03:
ist schon gefixt wenn die 0.9.x raus kommt gehts wieder :)

slg60, 2025-02-18:
> > ist schon gefixt wenn die 0.9.x raus kommt gehts wieder :)
> 
> Interessant! Wann können wir "Wartenden" denn mit der neuen Version rechnen? OT: Ich hoffe auch noch immer auf einen Fix für die sporadischen Reboots (besonders bei aktiviertem MQTT).

Das würde mich auch interessieren, wird die Ahoy nicht mehr weiterentwickelt? Seit Oktober gab es kein Update mehr. Bei Open DTU schon.

Dirk-Ahoy, 2025-02-18:
Die Entwicklung findet gerade in kleinerer Gruppe statt -  leider ist die "Auffrischung" doch sehr umfangreich geworden und es ist ja ein Freizeitprojekt  wo nur wenige dran arbeiten.   Also grössere Baustelle geworden und viel mehr Arbeit als gedacht.  Wir basteln aber fleissig im Kämmerlein :)

slg60, 2025-02-18:
Ah gut zu wissen. Freut mich das es weiter geht. Wenn ihr Tester braucht meldet euch gerne. Gibt es auch eine Lösung für die sporadischen Neustarts?

dknubben, 2025-02-18:
> Die Entwicklung findet gerade in kleinerer Gruppe statt - leider ist die "Auffrischung" doch sehr umfangreich geworden und es ist ja ein Freizeitprojekt wo nur wenige dran arbeiten. Also grössere Baustelle geworden und viel mehr Arbeit als gedacht. Wir basteln aber fleissig im Kämmerlein :)

Das zu hören freut mich natürlich grundsätzlich. Und es ist auch klar, dass hier einige (wenige) ihre wertvolle Freizeit opfern. Allerdings ist es doch gerade Sinn und Zweck eines GitHub-Projekts, möglichst viele einzubinden und die Feedbacks und Requests bei der Entwicklung zu nutzen und sich die Sache leichter zu machen. Es gibt sicher genügend Leute inkl. mir, die sich in verschiedener Form einbringen würden. Wäre es nicht auch besser, mit "Milestones" auf dem DEV-Branch zu arbeiten und so zu versuchen, neue Fehlerquellen zu vermeiden?