lumapu/ahoy #1803: Mit welchen Befehlen kann ich über die Webkonsole z.B. den Accesspoint aktivieren

RoiderTom, 2025-02-26:
## Hardware
  * [ x] ESP8266
  
### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
 
### Antenna:
* [x ] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
 
 ### Build & Flash Method:
* [ x] AhoyDTU Webinstaller
 
Ursache: das WLAN wurde geändert ohne den AhoyDTU vorher anzupassen. Nun spannt der AHOY keinen Accesspoint auf, damit ich es umkonfigurieren kann. Ich würde mir gern sparen, es neu zu konfigurieren.
Deshalb greife über  https://ahoydtu.de/web_install/ auf die Webconsole zu.

Kommunikation findet auch statt, aber ich finde irgendwie nirgends eine Übersicht der möglichen Befehle für diese serielle konsole.
Ich würde halt gern den Acesspoint manuell starten oder alternativ auch einfach das WIFI neu setzen - habt ihr irgendwo eine Übersicht der möglichen Befehle?

Vielen Dank schon mal!

knickohr, 2025-02-26:
Gar nicht. Der AP sollte sich öffnen wenn die DTU sich nicht mit einem WiFi verbinden kann.

Vielleicht WiFi mal komplett abschalten. Dann sollte der AP definitiv funktionieren.

jimknopf63, 2025-02-27:
Beim 8266 hatte ich das gleiche Problem, falsche Firmware aufgespielt und kein Zugriff mehr. In einem Arduino Forum habe ich den Tip bekommen den 8266 7x an Strom und direkt wieder auszuschalten. Beim 8. einstöpseln spannt er wieder sein AccessPoint Netzwerk auf. Hat funktioniert und ich konnte die korrekte Firmware Flaschen das WLAN wieder einstellen.

RoiderTom, 2025-02-28:
Danke für die Infos!

Das mit Strom an/aus hat bei mir leider nicht geklappt. Ich habe jetzt neu installiert, jetzt gehts wieder.
Trotzdem - Ahoy ist ein tolles Projekt! Top!