lumapu/ahoy #1805: Nur noch 1 von 3 WR online

SnakeGER, 2025-03-07:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.153

### Github Hash

00000

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

0000

### Debug Serial Log output

```shell

```

### Error description

Hallo zusammen.
Nach dem Update auf die 153 wird die Verbindung über das cmt-modul nur noch zu einem von 3 Wechselrichtern hergestellt. Die anderen beiden lassen sich weder auslesen, noch steuern. Auch ein Neustart der DTU brachte keine Hilfe. Nach dem Downgrade auf die 152, stimmte die Kommunikation sofort wieder.
In den Development Changes von der 153 steht auch etwas von "improved CMT communication".
Ich denke, hier wurde mal wieder etwas kaputt-repariert.
LG

lumapu, 2025-03-07:
Sehr schade zu hören, ich wollte eigentlich was gutes tun, habe es dabei aber wohl schlechter gemacht als zuvor. 
Ein System mit 2 Wechelrichtern lief mit dieser Version und hatte mit beiden Kontakt.
Hast du evtl. ein Log für mich, damit ich mir das näher ansehen kann?

SnakeGER, 2025-03-08:
Der Fehler scheint sporadisch aufzutreten. Ich hatte gerade die 153 installiert und es lief einwandfrei. Als ich die 153 das letzte Mal installiert hatte, lief auch alles, bis zum nächsten Morgen. Da war dann nur noch ein WR ansprechbar, was auch ein Neustart der DTU nicht beheben konnte.

mroenne2022, 2025-03-09:
Kurze Frage in dem Zusammenhang : Die Version 154 war kurz "online", nun nicht mehr. Ist die 154 wegen diesen hier beschriebenen Problemen gestoppt worden ?

lumapu, 2025-03-09:
nein, ich habe vergessen für das Release den dev-branch in main zu mergen. D.h. die dev-0.8.153 sollte Release werden, das Release war aber letzendlich eine 0.8.140 mit geänderter Versionsnummer.
Ich werde gleich die 0.8.155 auf den Weg bringen, die dann mein ursprüngliches Vorhaben umsetzt.

slg60, 2025-03-13:
Habe auch das Phänomen mit der 155, dass einer der 2 HMS1600 sich verabschiedet und auch nicht wieder kommt. Neustart des Inverters, keine Abhilfe, Neustart der DTU, keine Abhilfe, Downgrade auf meine 127 (einzige Version die bei mir fast reibungslos läuft) beide Inverter wieder da....

lumapu, 2025-03-13:
hmmm 🤔, dann habe ich aus eurer Sicht keine Verbesserung eingebaut, während es bei mir mit 4 HMS Wechselrichtern Verbesserung gezeigt hat

DasQwarx, 2025-04-10:
hi, ich habe selbiges problem. und ich kann die verbindung "kurzzeitig" damit wiederherstellen wenn ich in "einstellungen->wechselrichter->den verschwundenen wechselrichter BEARBEITE->funkmodul->"Wechselrichter suchen" klicke. 
I.d.R taucht er dann wieder auf ... bis willkührlich später ein anderer verschwindet.

zu mir ich betreibe zwei ahoy-dtu, eine mit beiden funkmodulen und eine mit nur dem cmt modul. der fehler tritt bei beiden DTU`s auf. Und auch erst seit ich von 140 auf neuere geupdatet hab. akuell hab ich 155 drauf. werd aber wieder zurück flashen.

knickohr, 2025-06-03:
Nun mal langsam Leute, beruhigt euch !

Korrekt, die 0.8.x wird nicht mehr weiterentwickelt. Es wird seit gut einem Jahr, an einer 0.9 programmiert. Leider sind in dieser Version noch viele Bugs drin, die ausgemerzt werden müssen. Da es sich um eine komplett neue Ahoy handelt, die fast komplett anders funktioniert, sprich neu aufgesetzt wurde, ist das kein einfaches Unterfangen. Diese neue 0.9 beinhaltet Features, die mit der 0.8 so nicht umsetzbar waren.

[image]
[image]
[image]
[image]

Habt Geduld, gut Ding will Weile haben 😉

knickohr, 2025-06-03:
Und das Problem mit den „verloren gehenden“ Schäfchen HMS ist bekannt und wird in der 0.9 behoben sein.

knickohr, 2025-06-03:
Ihr werdet nicht dumm sterben, zumindest nicht ab den Zeitpunkt wo es keine Alpha mehr ist und die ersten Betatester drauf los gelassen werden. Dann darfst Du auch gerne Moderator für die Betatester werden und deren Probleme und Fragen moderieren 😅

Scherz beiseite, aber wir haben momentan selbst genügend Issues mit den Alphatester. Irgendwann müssen wir weiterprogrammieren und Bugs fixen anstatt eine große Community drauf los zu lassen die dann sich vor eventuellen Problemen nicht mehr retten können.

Wir müssen erst unsere Hausaufgaben machen,