lumapu/ahoy #1807: Build for esp8266/esp8285 failed

MaDill, 2025-03-09:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.154

### Github Hash

c2e5d09

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

VS Studio Code 1.98.0
Platform.io 3.3.4
Folder is github checkout

### Debug Serial Log output

```shell

```

### Error description

1) checkout the github project in an empty folder
2) open the workspace configuration file
3) wait till platform.io prepared all the stuff
4) patch scripts/auto_firmware_version.py as in the https://github.com/lumapu/ahoy/pull/1759
5) do not select any specific environment; keep "Default (src)"
6) press "PlatformIO: Build"
7) on all esp8266 and esp8285 I receive more or less these errors

```
Processing esp8266-all (platform: espressif8266; board: esp12e; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: gc2e5d09
'../patches/AsyncWeb_Prometheus.patch' already applied
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2024.9.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.7.8
 - tool-esptoolpy @ 4.7.6
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.3.0+sha.c76c982
|-- RF24 @ 1.4.8+sha.638095e
|-- Time @ 1.6.1
|-- espMqttClient @ 1.7.0+sha.70d3113
|-- ArduinoJson @ 6.21.5
|-- Timezone @ 1.2.4+sha.7cf1425
|-- U8g2 @ 2.36.5
|-- GxEPD2 @ 1.5.3+sha.17cf7f4
|-- ESPAsyncUDP @ 0.0.0+sha.697c75a
|-- DNSServer @ 1.1.1
|-- ESP8266mDNS @ 1.2
|-- ESP8266WiFi @ 1.0
|-- Adafruit GFX Library @ 1.12.0
|-- SPI @ 1.0
|-- LittleFS(esp8266) @ 0.1.0
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- Hash @ 1.0
|-- Adafruit BusIO @ 1.17.0
|-- Wire @ 1.0
Building in release mode
Compiling .pio\build\esp8266-all\src\app.cpp.o
Compiling .pio\build\esp8266-all\src\main.cpp.o
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
Compiling .pio\build\esp8266-all\src\network\AhoyNetworkHelper.cpp.o
Compiling .pio\build\esp8266-all\src\plugins\Display\Display_ePaper.cpp.o
*** [.pio\build\esp8266-all\src\app.cpp.o] Error 1
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
*** [.pio\build\esp8266-all\src\network\AhoyNetworkHelper.cpp.o] Error 1
*** [.pio\build\esp8266-all\src\plugins\Display\Display_ePaper.cpp.o] Error 1
*** [.pio\build\esp8266-all\src\main.cpp.o] Error 1
```

8) This is the final result
```

Environment                  Status    Duration
---------------------------  --------  ------------
esp8266-minimal              FAILED    00:00:40.682
esp8266                      FAILED    00:00:27.576
esp8266-de                   FAILED    00:00:25.541
esp8266-all                  FAILED    00:00:26.558
esp8266-all-de               FAILED    00:00:27.286
esp8266-prometheus           FAILED    00:00:28.097
esp8266-prometheus-de        FAILED    00:00:23.931
esp8285                      FAILED    00:00:23.017
esp8285-de                   FAILED    00:00:25.197
esp32-wroom32-minimal        SUCCESS   00:01:42.702
esp32-wroom32                SUCCESS   00:02:02.169
esp32-wroom32-de             SUCCESS   00:02:04.807
esp32-wroom32-prometheus     SUCCESS   00:02:09.342
esp32-wroom32-prometheus-de  SUCCESS   00:01:50.669
esp32-s2-mini                SUCCESS   00:02:03.521
esp32-s2-mini-de             SUCCESS   00:01:45.131
esp32-c3-mini                SUCCESS   00:01:27.916
esp32-c3-mini-de             SUCCESS   00:01:25.029
opendtufusion-minimal        SUCCESS   00:02:37.082
opendtufusion                SUCCESS   00:02:50.491
opendtufusion-de             SUCCESS   00:02:12.112
opendtufusion-16MB           SUCCESS   00:01:49.701
opendtufusion-16MB-de        SUCCESS   00:01:40.306
====================================================== 9 failed, 14 succeeded in 00:31:48.862 ====================================================== 

```

lumapu, 2025-03-09:
feels invalid, because github action can build the ESP8266 without any problem:
https://github.com/lumapu/ahoy/actions/runs/13752088883

knickohr, 2025-03-09:
Please don’t use 154. This Version is buggy.

MaDill, 2025-03-09:
I just updated to https://github.com/lumapu/ahoy/commit/5feb293c9f53e980e591a0ec9b8851794a011743 and tried again. Then I compared the output of the github action https://github.com/lumapu/ahoy/actions/runs/13752088883/job/38454332732 with my one.

This come from the github action:
```
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
```

This show up on my computer:
```
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.7.8 
 - tool-esptoolpy @ 4.7.6
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
```

Where does this versions come from? What do I have to install?

MaDill, 2025-03-10:
I found the issue on my computer. I have also https://github.com/arendst/Tasmota/ in my VS Studio code. This link an own framework from https://github.com/tasmota/platform-espressif8266. This has the text version number 2.7.8, but reported as 2024.9.0 with the same name as the original espressif8266 platform.

Then the version check acording to semver decide 2024.9.0 is newer than 4.2.1 from https://registry.platformio.org/platforms/platformio/espressif8266.

The final think is the platformio.ini linked "platform = espressif8266" and this use the latest version. Now I changed this to "platform = espressif8266 @ ^4.2.1" and the build is working.

I will file a report on https://github.com/tasmota/platform-espressif8266

closed 2025-03-10 by MaDill