lumapu/ahoy #1808: Ahoy reboot nach MQTT Verbindungsabbruch

Krausbirn, 2025-03-10:
### Platform

ESP32

### Assembly

I did the assembly by myself

### nRF24L01+ Module

round dot

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.155

### Github Hash

5feb293

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

standard + MQTT Broker set up

### Debug Serial Log output

```shell
08:23:35.002 I: (#0) Radio infos: 0 4 -2 0 0 | t: 52, s: 52, f: 0, n: 0 | p: 0
08:23:35.005 I: (#0) TX 27 CH23, 9 ret. | 15 0b 80
08:23:35.170 I: (#0) RX  32ms | 27 CH61 | 95 01
08:23:35.171 I: (#0) RX  70ms | 27 CH61 | 95 02
08:23:35.172 I: (#0) RX 117ms | 23 CH40 | 95 83
08:23:35.173 I: (#0) Payload (42)
08:23:35.176 -----
08:23:35.177 I: com loop duration: 176ms
08:23:35.177 -----
08:23:45.002 I: (#0) Radio infos: 0 4 -2 0 0 | t: 53, s: 53, f: 0, n: 0 | p: 0
08:23:45.005 I: (#0) TX 27 CH23, 9 ret. | 15 0b 80
08:23:45.182 I: (#0) RX  36ms | 27 CH75 | 95 01
08:23:45.183 I: (#0) RX  82ms | 27 CH61 | 95 02
08:23:45.184 I: (#0) RX 130ms | 23 CH61 | 95 83
08:23:45.185 I: (#0) Payload (42)
08:23:45.188 -----
08:23:45.189 I: com loop duration: 188ms
08:23:45.189 -----
00:00:09.010 I: MQTT disconnected, reason: TCP disconnect
00:00:13.011 I: MQTT disconnected, reason: TCP disconnect
00:00:15.003 I: (#0) Radio infos: 0 0 0 0 0 | t: 0, s: 0, f: 0, n: 0 | p: 0
00:00:15.005 I: (#0) TX 27 CH23, 15 ret. | 15 0b 80
00:00:15.170 I: (#0) RX  40ms | 27 CH75 | 95 01
00:00:15.171 I: (#0) RX  68ms | 27 CH61 | 95 02
00:00:15.171 I: (#0) RX 118ms | 23 CH61 | 95 83
00:00:15.172 I: (#0) Payload (42)
00:00:15.173 I: alarm ID incremented to 19
00:00:15.175 -----
00:00:15.175 I: com loop duration: 173ms
00:00:15.186 -----
00:00:17.011 I: MQTT disconnected, reason: TCP disconnect
00:00:21.012 I: MQTT disconnected, reason: TCP disconnect
00:00:25.001 I: (#0) Radio infos: 0 2 0 0 0 | t: 1, s: 1, f: 0, n: 0 | p: 0
00:00:25.004 I: (#0) TX 27 CH23, 12 ret. | 15 0b 80
00:00:25.010 I: MQTT disconnected, reason: TCP disconnect
00:00:25.179 I: (#0) RX  38ms | 27 CH61 | 95 01
00:00:25.180 I: (#0) RX  79ms | 27 CH61 | 95 02
00:00:25.180 I: (#0) RX 128ms | 23 CH40 | 95 83
00:00:25.181 I: (#0) Payload (42)
00:00:25.182 I: alarm ID incremented to 20
00:00:25.184 -----
00:00:25.184 I: com loop duration: 183ms
00:00:25.195 -----
00:00:29.011 I: MQTT disconnected, reason: TCP disconnect
00:00:33.012 I: MQTT disconnected, reason: TCP disconnect
00:00:35.002 I: (#0) Radio infos: 0 4 0 0 0 | t: 2, s: 2, f: 0, n: 0 | p: 0
00:00:35.004 I: (#0) TX 27 CH23, 9 ret. | 15 05 80
00:00:35.077 I: (#0) RX  25ms | 27 CH75 | 95 81
00:00:35.078 I: (#0) Payload (14)
00:00:35.080 -----
00:00:35.081 I: (#0) Radio infos: 0 4 0 0 0 | t: 3, s: 3, f: 0, n: 0 | p: 0
00:00:35.082 I: (#0) TX 27 CH23, 9 ret. | 15 0b 80
00:00:35.246 I: (#0) RX  32ms | 27 CH61 | 95 01
00:00:35.247 I: (#0) RX  70ms | 27 CH61 | 95 02
00:00:35.247 I: (#0) RX 117ms | 23 CH40 | 95 83
00:00:35.248 I: (#0) Payload (42)
00:00:35.250 -----
00:00:35.251 I: com loop duration: 250ms
00:00:35.251 -----
00:00:37.010 I: MQTT disconnected, reason: TCP disconnect
```

### Error description

Verbingsabbruch vom MQTT Broker führt zu einem Ahoy restart. 
Die Versionen 0.8.152, 0.8.153, 0.8.155 verhalten sich gleich 

Gerätename	AHOY-DTU
Laufzeit	0 days, 00:01:42
Grund des Neustarts	Task Watchdog
Umgebung	esp32-wroom32-de (Module: MDH-de)
Version	0.8.155 - 5feb293
Chip	CPU: 240MHz, 2 Core(s)
Chip Model	ESP32-D0WD-V3

lumapu, 2025-03-13:
du hast aber gleich mehrere Probleme, wenn die Alarm ID so schnell hoch zählt. Ich will damit deine These nicht entkräften, aber das sticht ins Auge.

Krausbirn, 2025-03-13:
Das System läuft an sich sehr stabil. Der Fehler ist auch ganz leicht nachstellbar. Bei dem Log hab mehrmals getestet. Hin und wieder, wenn der Broker nur ganz kurz weg ist dann überstehts der Ahoy auch ohne reboot. 
Warum ist mir das Verhalten aufgefallen, mein Broker läuft auf einem ESP32 und oft wenn ich ein OTA Update vom Broker mache bootet auch Ahoy neu. Andere ESP Clients die bei mir auch über MQTT mit dem Broker kommunizieren überstehens problemlos. Kann leider nicht sagen seit welcher Version sich das Problem eingeschlichen hat, mit älteren Versionen hatte ich da nie Probleme.

Krausbirn, 2025-03-17:
Hab jetzt mal einige ältere Versionen probiert. 0.8.140, 0.8.145, 0.8.150 gleiches Verhalten.

Was mir aber aufgefallen ist mit der 0.8.153 wenn keine Verbindung zum WR besteht, dann läuft der AHOY ohne reboot weiter.

[1.txt](https://github.com/user-attachments/files/19282929/1.txt)