lumapu/ahoy #1809: Frage zu Discord

Loetnase, 2025-03-10:
@knickohr
Wird irgendwas unter meinem Namen drüben im Discord gemacht?
Ich hatte vorhin E-Mails mit einem Passwortrücksetzung Discord in meinem E-mail Eingang welche nach ein paar Minuten wieder verschwunden waren. 
Im E-Mail Papierkorb sind sie auch nicht mehr gewesen. Komme jetzt nicht mehr in discord rein.
Ich denke da hat jemand meinen E-Mail Account gehackt und war da aktiv drauf. 
Da habe ich das Passwort schon gewechselt

In meinen Discord-Account komme ich nicht mehr rein auch mit Passwortrücksetzung geht nichts.
Hat mit jemand einen Tipp wie ich da wieder rein- oder rankomme
oder muss/kann ich mich nur neu anmelden?

Danke

knickohr, 2025-03-10:
Versuche irgendwie in Deinen Account zu kommen. Alles andere was ich hier schreibe bekommt der böse Bub auch mit da er wahrscheinlich auch Deine eMail gekapert hat. Generell, alle Dir möglichen PWs ändern, aber nicht bevor Du Deinen PC mit einem guten Virenscanner gecleant hast. Er könnte Dir einen Trojaner untergeschoben haben und somit wäre alle Mühe mit den PWs umsonst 😥

DanielR92, 2025-03-10:
Richtig,

ich habe die Info bekommen das es Probleme mit deinem Account gab,
Zur Sicherheit habe ich die Rechte entzogen und ein TimeOut gesetzt. 

Da ich nicht sicher nun bin, das der richtige User nun sein Account wieder hat. 
Bitte verifziere dich vielleicht mittels 2FA.

Dein Handy hatte mitgeteilt das eine Email von Discord angekommen ist. Wurde aber wieder gelöscht.
Also hat jemand auch Zugriff auf deine Emails gehabt. Sowas ist immer sehr kritisch zu sehen.

Vergewissere dich, das ein neues Passwort überall gesetzt worden ist. 
Wenn alles wieder passt, einfach hier melden und ich entferne den Timeout im voraus und gebe dir die Rechte wieder.

Gruß

Loetnase, 2025-03-12:
@DanielR92 
Hallo Daniel,
über den Support von discord wurde ich wieder freigeschalten, das Passwort ist geändert und ich werde jetzt 2FA einrichten. 
Das habe ich beim E-mail Account und anderen wichtigen Seiten auch schon gemacht. Das ist aber auch schon erstaunlich wie viele unterschiedliche Verfahren bei 2FA da genutzt werden. Wenn da mein Rechner mal richtig hops geht dann  wird das interesannt wie ich das wieder alles ans Laufen bekomme.
Ich bin also wieder da, kannst du mich wieder direkt freischalten oder brauchst du irgendwas um zu sehen das ich auch wirklich die Loetnase bin.

Danke

Loetnase, 2025-03-13:
Ich bin jetzt im Discord wieder drin.
Danke für die Hilfe und Antworten, 
das Issue mach e ich auch gleich zu.
:pray:

closed 2025-03-13 by Loetnase