lumapu/ahoy #1810: Installation auf ESP32 S3 nicht möglich!?

huaak, 2025-03-12:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32 S3
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
Beim Versuch auf einen ESP32 S3 Ahoy aufzuspielen kommt (unabhängig mit welcher der beiden USB-C-Ports ich es probiere) nur die Fehlermeldung dass der S3 nicht supported sei..ist das normal, habe ich einen komischen ESP oder gibt es einen Trick den ich verpasst habe?!

[image]

knickohr, 2025-03-12:
Das geht mit dem Webflasher nicht.

Nimm ESP Downloadtool von espressif, oder den IDF, oder über PlattfotmIO.

Gubi2023, 2025-03-12:
versuchs mal damit:  https://fw.ahoydtu.de/tools/webflasher/

huaak, 2025-03-12:
> versuchs mal damit: https://fw.ahoydtu.de/tools/webflasher/

Super, danke-da hat die Installation funktioniert! :)

whadd, 2025-03-12:
Hi, ich hatte das gleiche Problem...

Meine Lösung war simpy ...  flashen einer alten Version von AHOY - DTU ( ich glaube es war 0.8.83 )
und dann alle Installationsdaten von meinen 3 Hoymiles und Netzwerkzugang etc.  korrekt eingeben...
 Dritter Schritt: Herunterladen der 0.8.155 - Datei    und dann  "update"   aus dem  AHOY - Menü

da hatte ich eben die hier oben vorgeschlagenen Tips  noch nicht gesehen ... hat jedenfalls funktioniert.