lumapu/ahoy #1811: Can not connect to Inverter HMS-800 with v0.8.155

outermedia-cbo, 2025-03-12:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.155

### Github Hash

5feb293

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

CMT2300A is configured correctly:

[image]

### Debug Serial Log output

```shell
webSerial, buffer overflow!
14:08:19.236 (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 0b 00 67 d1 87 43 00 00 00 00 00 00 00 00 bf eb 90 
14:08:19.281 W: (#0) frame 1 missing: request retransmit (3 attempts left)
14:08:19.282 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 81 bc 
14:08:19.317 W: (#0) frame 1 missing: request retransmit (2 attempts left)
14:08:19.318 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 81 bc 
14:08:19.357 W: (#0) frame 1 missing: request retransmit (1 attempts left)
14:08:19.358 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 81 bc 
14:08:19.388 W: (#0) frame 1 missing: request retransmit (0 attempts left)
14:08:19.389 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 81 bc 
14:08:19.431 I: (#0) timeout, no attempts left
14:08:19.432 -----
14:08:19.432 I: com loop duration: 433ms
14:08:19.433 -----
X 297ms | 27 -59dBm | 95  *  *  *  *  *  *  *  * 07 00 00 08 5b 01 2c 08 b7 09 41 09 9d 01 2c 00 64 ee 
14:08:19.113 I: (#0) RX 347ms | 27 -59dBm | 95  *  *  *  *  *  *  *  * 08 90 00 00 01 00 5f b0 00 00 00 01 f4 00 5f 70 02 13 
14:08:19.127 I: (#0) RX 394ms | 23 -59dBm | 95  *  *  *  *  *  *  *  * 89 00 01 27 10 a0 02 00 00 00 00 f6 03 55 
14:08:19.132 W: (#0) frame 1 missing: request retransmit (6 attempts left)
14:08:19.133 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 81 bc 
14:08:19.187 I: (#0) RX 122ms | 27 -59dBm | 95  *  *  *  *  *  *  *  * 01 0d 04 20 00 00 0c 08 fc 07 a3 00 0f 09 e2 00 1e 33 
14:08:19.191 I: (#0) Payload (138): 0d 04 20 00 00 0c 08 fc 07 a3 00 0f 09 e2 00 1e 06 4a 00 14 0a 55 00 14 0a c8 00 0a 09 e2 10 03 13 88 12 c0 00 14 13 ec 00 14 12 8e 00 32 14 50 00 32 20 00 00 01 30 03 02 58 09 e2 07 a3 13 9c 13 56 40 00 07 d0 00 10 50 01 00 01 13 9c 01 90 00 10 00 00 60 00 00 01 09 e2 0a 5a 02 15 80 01 00 00 08 5b 01 2c 08 b7 09 41 09 9d 01 2c 00 64 90 00 00 01 00 5f b0 00 00 00 01 f4 00 5f 70 02 00 01 27 10 a0 02 00 00 00 00 
14:08:19.229 -----
14:08:19.230 I: (#0) Radio infos: t: 115, s: 4, f: 0, n: 111 | p: 20
14:08:24.000 I: (#0) Radio infos: t: 116, s: 4, f: 0, n: 112 | p: 20
14:08:24.003 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 0b 00 67 d1 87 48 00 00 00 00 00 00 00 00 8f 98 d8 
14:08:24.024 I: (#0) RX  68ms | 27 -59dBm | 95  *  *  *  *  *  *  *  * 01 0d 04 20 00 00 0c 08 fc 07 a3 00 0f 09 e2 00 1e 33 
14:08:24.028 W: (#0) frame 2 missing: request retransmit (3 attempts left)
14:08:24.029 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 82 bf 
14:08:24.041 W: CMT TX failed, code: 2
14:08:24.044 W: (#0) frame 2 missing: request retransmit (2 attempts left)
14:08:24.045 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 82 bf 
14:08:24.073 W: (#0) frame 2 missing: request retransmit (1 attempts left)
14:08:24.074 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 82 bf 
14:08:24.096 W: (#0) frame 2 missing: request retransmit (0 attempts left)
14:08:24.097 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 82 bf 
14:08:24.099 W: CMT TX failed, code: 2
14:08:24.138 I: (#0) timeout, no attempts left
14:08:24.139 -----
14:08:24.139 I: (#0) Radio infos: t: 117, s: 4, f: 0, n: 113 | p: 20
14:08:24.141 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 15 00 67 d1 87 48 00 00 00 00 00 00 00 00 91 86 c6 
14:08:24.192 I: (#0) request timeout: 2ms
14:08:24.193 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
14:08:24.196 -----
14:08:24.197 I: com loop duration: 197ms
14:08:24.198 -----
14:08:29.010 I: (#0) Radio infos: t: 118, s: 4, f: 0, n: 114 | p: 20
14:08:29.013 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 0b 00 67 d1 87 4d 00 00 00 00 00 00 00 00 df a7 b2 
14:08:29.069 I: (#0) request timeout: 2ms
14:08:29.070 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
14:08:29.073 -----
14:08:29.074 I: com loop duration: 65ms
14:08:29.074 -----
```

### Error description

I upgraded to last stable: v0.8.155

Now it does not connect to HMS-800. I waited for several hours. 

There some `webSerial, buffer overflow!`. I guess that means there is too much data to be logged. Or is this the cause?

I rather assume `W: CMT TX failed, code: 2` is hte problem.

Connection with v0.8.152 was OK before upgrade and is OK again after downgrade.

Gubi2023, 2025-03-12:
have you tried power off/on?

outermedia-cbo, 2025-03-12:
No, just restarts, yet. I am going to do it.

outermedia-cbo, 2025-03-12:
No more errors after turning it off and on again, but no connection either:

```
15:41:07.150 -----
15:41:12.001 I: (#0) Radio infos: t: 232, s: 1, f: 0, n: 231 | p: 20
15:41:12.003 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 01 00 67 d1 9d 08 00 00 00 00 00 00 00 00 f3 82 ee 
15:41:12.033 I: (#0) request timeout: 2ms
15:41:12.034 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
15:41:12.058 -----
15:41:12.059 I: (#0) Radio infos: t: 233, s: 1, f: 0, n: 232 | p: 20
15:41:12.061 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 0b 00 67 d1 9d 08 00 00 00 00 00 00 00 00 f9 88 e4 
15:41:12.157 I: (#0) request timeout: 2ms
15:41:12.158 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
15:41:12.161 -----
15:41:12.162 I: com loop duration: 163ms
15:41:12.163 -----
15:41:17.001 I: (#0) Radio infos: t: 234, s: 1, f: 0, n: 233 | p: 20
15:41:17.003 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 01 00 67 d1 9d 0d 00 00 00 00 00 00 00 00 a3 bd 84 
15:41:17.062 I: (#0) request timeout: 2ms
15:41:17.063 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
15:41:17.067 -----
15:41:17.067 I: (#0) Radio infos: t: 235, s: 1, f: 0, n: 234 | p: 20
15:41:17.069 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 0b 00 67 d1 9d 0d 00 00 00 00 00 00 00 00 a9 b7 8e 
15:41:17.123 I: (#0) request timeout: 2ms
15:41:17.124 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
15:41:17.158 -----
15:41:17.159 I: com loop duration: 159ms
15:41:17.160 -----
15:41:22.001 I: (#0) Radio infos: t: 236, s: 1, f: 0, n: 235 | p: 20
15:41:22.004 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 01 00 67 d1 9d 12 00 00 00 00 00 00 00 00 93 31 27 
15:41:22.053 I: (#0) request timeout: 2ms
15:41:22.054 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
15:41:22.058 -----
15:41:22.059 I: (#0) Radio infos: t: 237, s: 1, f: 0, n: 236 | p: 20
15:41:22.060 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 0b 00 67 d1 9d 12 00 00 00 00 00 00 00 00 99 3b 2d 
15:41:22.153 I: (#0) request timeout: 2ms
15:41:22.154 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
15:41:22.157 -----
15:41:22.158 I: com loop duration: 159ms
15:41:22.159 -----
15:41:27.011 I: (#0) Radio infos: t: 238, s: 1, f: 0, n: 237 | p: 20
15:41:27.015 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 01 00 67 d1 9d 17 00 00 00 00 00 00 00 00 c3 0e 4d 
15:41:27.069 I: (#0) request timeout: 0ms
15:41:27.070 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
15:41:27.074 -----
15:41:27.075 I: (#0) Radio infos: t: 239, s: 1, f: 0, n: 238 | p: 20
15:41:27.077 I: (#0) TX 863.00Mhz | 15  *  *  *  * 84 16 06 10 80 0b 00 67 d1 9d 17 00 00 00 00 00 00 00 00 c9 04 47 
15:41:27.167 I: (#0) request timeout: 2ms
15:41:27.167 I: (#0) TX 868.00Mhz | 56  *  *  *  * 84 16 06 10 02 15 21 0c 14 2d 51 2c 
15:41:27.171 -----
15:41:27.172 I: com loop duration: 161ms
15:41:27.173 -----
```

Gubi2023, 2025-03-12:
tja, die DTU sieht den WR nicht mehr, warum auch immer... (Standort geändert?) Blinkt der WR grün?

dawid-woitaschek, 2025-03-12:
Same here...

Kann ich einfach die 153 draufbrezeln oder stellt sich was einem Downgrade in die Quere?

Gubi2023, 2025-03-12:
nö, wobei ich mit der .153 vorsichtig wäre, evtl. buggy

outermedia-cbo, 2025-03-12:
> tja, die DTU sieht den WR nicht mehr, warum auch immer... (Standort geändert?) Blinkt der WR grün?

Standort von DTU und Wechselrichter sind gleich. Mit der 152er Firmware geht es ja auch. 
Der WR blinkt tagsüber grün.

lumapu, 2025-03-13:
echt eigenartig, kann mir noch nicht erklären woher das kommen könnte.
Sieht die Situation am nächsten Morgen evtl. besser aus? Dazu einfach nachts das Update machen und schauen, ob es morgens wie gewünscht läuft.

outermedia-cbo, 2025-03-17:
@lumapu Das habe ich über das Wochenende auch probiert: Keine Verbindung.

Die Version 152 lief bei mir sehr stabil. Ich werde sie weiterverwenden.

Gubi2023, 2025-03-18:
merkwürdig: hatte gestern auch das Problem, dass die DTU mit .155 den WR ab 16Uhr nicht mehr sah, mit .152 gehts...

busaku, 2025-03-28:
Hat hier noch jemand die .152 ?
Finde die zum Download nicht mehr und ich habe mit 153 und 155 massive Verbindungsabbrüche. Der Inverter ist alle paar Minuten/Stunden weg.

Es hilft zwar in in den Settings neu zu Catchen, aber da ich einige Automatisierungen damit laufen habe ist das eine eher weniger nachhaltige  Lösung :D

blueline13, 2025-03-28:
Moin,

die FW findest du hier: https://fw.ahoydtu.de/fw/dev%2Fahoy_v0.8.152/

Mit 155 war heute Morgen eine meiner Invert auch nicht verfügbar trotz Reboot. Nach zwei Tagen testen bin ich auch wieder zurück auf die 152.

busaku, 2025-03-28:
Danke, allerdings scheint es bei mir vielleicht was anderes zu sein. Mit 152 bekomme ich nur noch einmal nach dem Reboot eine kurze Verbindung, dann wars das direkt auch schon wieder.

`09:24:05.139 -----
09:24:05.141 W: CMT TX failed, code: 2
09:24:05.498 I: (#0) RX  68ms | 27 -39dBm | 95 81
09:24:05.500 -----
09:24:05.500 I: com loop duration: 499ms
09:24:05.501 -----
09:24:20.138 -----
09:24:20.139 W: CMT TX failed, code: 2
09:24:20.499 I: (#0) RX  71ms | 27 -39dBm | 95 81
09:24:20.501 -----
09:24:20.501 I: com loop duration: 502ms
09:24:20.502 -----
09:24:35.151 -----
09:24:35.153 W: CMT TX failed, code: 2
09:24:35.511 I: (#0) RX  68ms | 27 -39dBm | 95 81
09:24:35.513 -----
09:24:35.514 I: com loop duration: 503ms
09:24:35.516 -----
`

Gubi2023, 2025-03-28:
bei mir läuft die .152 inzwischen seit 31 Tagen problemlos! Die 155 lief nicht durch....

Jan-x-28, 2025-03-31:
Same to me since this morning. MfG Jan

sandboxgangster, 2025-06-11:
Hab das Problem leider auch, allerdings nur Stundenweise hab zwei HM-300 beide sind direkt untereinander dachte erst das liegt am Möbel umbau und daher Empfangsprobleme, ist es allerdings nicht DTU direkt neben die WR gepackt hilft nix.

Was mich wundert es betrifft nur den ersten WR, der zweite geht Problemlos.