lumapu/ahoy #1814: HMS 1600 4T - not possible to change power limit

gregors101, 2025-03-21:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

250305_ahoy_0.8.154_c2e5d09_opendtufusion-de.bin

### Github Hash

c2e5d09

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Nothing special

### Debug Serial Log output

```shell

```

### Error description

It is not possible to change the power limit, the command timed out. Also the power limit is not displyed

Dirk-Ahoy, 2025-03-21:
Welche Firmware ist am HMS1600 ?

knickohr, 2025-03-22:
Please update to the latent stable version (155).

gregors101, 2025-03-23:
@knickohr I tried with latest version 155, same behaviour as with version 154

gregors101, 2025-03-23:
@Dirk-Ahoy 
Hardware Version H00.04.10
Firmware Version V02.00.04

Sometimes it accepts the change without error, but it still does not work.

[image]

Dirk-Ahoy, 2025-03-23:
The Problem is the new F/W  2.xxx  

https://github.com/tbnobody/OpenDTU/discussions/2153#discussioncomment-11914884

stefan123t, 2025-05-18:
@Dirk-Ahoy and the `SystemConfigPara` Response also changed from version 1 to 3, this may lead to 100% shown on DTU while Limit is actually something else.

https://github.com/tbnobody/OpenDTU/issues/2641#issuecomment-2886433841

Zippochonda, 2025-07-04:
Will this also be fixed in Ahoy?

gregors101, 2025-08-25:
Is there a new state about this issue?

Dirk-Ahoy, 2025-08-25:
Downgrade the HMS to Version  1.1.12 -  it`s not a Problem from Ahoy - see here:

https://github.com/tbnobody/OpenDTU/issues/2784