lumapu/ahoy #1815: MQTT not connecting after update to 0.8.155

cgudrian, 2025-03-23:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### nRF24L01+ Module

round dot

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.155

### Github Hash

5feb293

### Build & Flash Method

ESP Tools (flash)

### Setup

- MQTT: Broker IP was set

### Debug Serial Log output

```shell
(no MQTT output in the Log)
```

### Error description

After update to 0.8.155 the MQTT connection is no longer established. The IP address of the broker is correctly specified in the interface. The broker is running (I'm connected to it via mosquitto_sub in parallel). The serial console shows no traces of MQTT activity at all.

cgudrian, 2025-03-23:
Forget it. I installed the "minimal" version which does include the MQTT settings but not the MQTT functionality itself.

closed 2025-03-23 by cgudrian

knickohr, 2025-03-23:
Which of the 4 versions for the 8266 have you installed ?

cgudrian, 2025-03-23:
Thanks for the quick reply. But I've already found out, that my stupid attempt to make MQTT work with the minimal build was futile. Hence I've closed this issue just a minute before your reply. 😄