lumapu/ahoy #1817: Wechselrichter Produktion bricht immer wieder ein nach Update auf 0.8.155 - sonst nichts verändert - Zufall?

SebiWolze, 2025-03-28:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.155

### Github Hash

-

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

2x Hoymiles HMS 800

### Debug Serial Log output

```shell

```

### Error description

Hello, I see many issue reports in german so I will write it in german as well :-)

Hallo,

wirklich keine Ahnung ob es einen Zusammenhang geben kann, ABER:
Ich habe heute von 0.8.152 auf die neueste Release Version geupdated und auf einmal stieg mein 2. Wechselrichter (Hoymiles HMS800) immer wieder aus. Watt Wert auf 0, keine Produktion mehr.
In einer Tour.
Dazu noch Grid Overcurrent -> laut Hoymiles liegt das an Widerstand/Kabellänge etc. Aber meine Setups (2x 800W Rigs, jeweils Hoymiles HMS800 WR) laufen so seit letztem Jahr Mai problemlos.

Ich habe NUR AhoyDTU geupdated, nichts eingestellt sonst, nichts verändert, Kabel sind gleich, Module gleich, ALLES gleich.

Der zeitliche Zusammenhang macht mich stutzig...
Habe dann wieder ein Downgrade vorgenommen, das Phänomen trat dann noch 1x auf, seitdem nicht mehr.
WR habe ich via AhoyDTU neugestartet.

Kann das irgendeinen Zusammenhang geben da ich ja durchaus via AhoyDTU Einstellungen am WR vornehmen kann (Max Wattage usw.) oder ist das absolut ausgeschlossen.

Gerne kontrolliere ich WR, Kabel usw. aber wie bereits erwähnt: Dieser zeitliche Zusammenhang macht mich vollkommen stutzig...

Jemand ähnliches erlebt evtl. sogar?

Danke!

Gubi2023, 2025-03-28:
https://github.com/lumapu/ahoy/issues/1811#issuecomment-2761687057

SebiWolze, 2025-03-28:
> [#1811 (comment)](https://github.com/lumapu/ahoy/issues/1811#issuecomment-2761687057)

Antwort ist da, Temp usw vom WR kann ich auslesen, live. Aber der WR produziert einfach nix mehr. Produktion 0 Watt.

Hoymiles1500, 2025-05-21:
Hallo, genau das gleiche Problem tritt bei meinen Wechselrichter Hoymiles HM1500 auch auf,
er zeigt mir ganz Normal die Werte und und plötzlich zeigt er NULL Watt an, obwohl eine Modullänge darüber ein 2 Wechselrichter gleicher Marke Größe hängt der da die anstehenden Werte korrekt anzeigt.
Nach einer unbestimmten Zeit zeigt er dann wieder Werte an.
Der Service von Hoymiles meinte auf meine Fragen diesbezüglich, ich solle mit eine DTU anschaffen damit die das auslesen können, möchte ich aber eigentlich nicht

Gubi2023, 2025-05-21:
Klick mal im "Daten"-Bild auf Meldungen, ob da was angezeigt wird...

Hoymiles1500, 2025-05-23:
Genau das gleiche Problem habe ich auch, ohne ersichtlichen Grund geht mein
Hoymiles WR 1500HM auch bei bestem Wetter plötzlich auf 0 Watt Produktion
und dann ganz plötzlich arbeitet er wieder
hab mich mit Hoymiles in Verbindung gesetzt, die meinten nur - ja wenn Sie
keine DTU haben, dann können wir da nicht reinschauen, kaufen Sie sich ne
DTU . das war´s.
Hatte dem Kundenservice angeboten den WR zu schicken, dann die
Überraschung, die haben gar keine Reparatur Abteilung !!!
Soviel zu Hoymiles die seit Jahren mit ihren Produkten prahlen und von
wegen Gewährleistung usw.
Wollte mir einen Hoymiles Batteriespeicher kaufen, aber das kommt gar nicht
mehr in Frage.