lumapu/ahoy #1818: 0.8.155 - DHCP Hostname No Set To DeviceName

gone-for-coding, 2025-03-30:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.155

### Github Hash

5feb293

### Build & Flash Method

ESP Tools (flash)

### Setup

System Config -> Device Host Name

### Debug Serial Log output

```shell

```

### Error description

Es scheint als sei dieser Bug https://github.com/lumapu/ahoy/issues/752 zurück. Ich baue gerade auf einem Fusion neu auf und habe daher aktuell keine DHCP-Reservierung. Ahoy sendet nicht den DeviceName sondern den default Namen im DHCP-Discover und landet dann leider auch so im DNS.

```
12:46:43.844505 IP (tos 0x0, ttl 64, id 0, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from e8:06:90:fc:27:80, length 308, xid 0x1c178c1a, Flags [none] (0x0000)
          Client-Ethernet-Address e8:06:90:fc:27:80
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message (53), length 1: Discover
            MSZ (57), length 2: 1500
            Hostname (12), length 14: "esp32s3-FC2780"
            Client-ID (61), length 7: ether e8:06:90:fc:27:80
            Parameter-Request (55), length 5:
              Subnet-Mask (1), Default-Gateway (3), BR (28), Domain-Name-Server (6)
              NTP (42)
```

Noch eine Info: Ich habe die 0.8.155 Factory von knickohr für das neue Partitionslayout aus dem Discord drauf.

gone-for-coding, 2025-04-04:
Ich habe ein bisschen gesucht, bin mir aber nicht sicher, ob ich den Code richtig verstehe. Daher geraten: Fehlt das Setzen des Hostnamen eventuell hier drunter?
https://github.com/lumapu/ahoy/blob/5feb293c9f53e980e591a0ec9b8851794a011743/src/network/AhoyNetwork.h#L31

Hier wird er gesetzt, aber da ist es ggf. schon zu spät?
https://github.com/lumapu/ahoy/blob/5feb293c9f53e980e591a0ec9b8851794a011743/src/network/AhoyWifiEsp32.h#L25

CDMartinL, 2025-04-09:
Ich habe das gleiche Problem, aber es scheint nicht bei jedem Boot so zu sein. Manchmal wird der richtige Name registriert, manchmal der default name (esp32s3-XXXXX).