lumapu/ahoy #1819: display not working

rabbisch, 2025-04-03:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2025-04-03:
Etwas ausführlicher bitte. Meine Glaskugel ist zur Zeit sehr trübe.

rabbisch, 2025-04-03:
AhoyDTU is working fine (V0.8.83)!  Thanks for all who contributed to this project !

I connected a  display (ssd 1306) and did the settings, 
but the display doesn't show anything?

Any idea: undervoltage? wrong settings? ...

[image]

[image]

knickohr, 2025-04-03:
SSD1306 or SH1106 ?

What Firmware do you use ?

[image]

closed 2025-04-04 by rabbisch