lumapu/ahoy #1820: 3. Wechselrichter HM400 - Authentifizierungsfehler

SebiWolze, 2025-04-03:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.155

### Github Hash

abc1234

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

2x HMS800
1x HM400

### Debug Serial Log output

```shell

```

### Error description

Hallo zusammen,

ich habe soeben versucht einen 3. Wechselrichter (HM400) in Betrieb zu nehmen.
Erhalte egal was ich eingebe/versuche immer "Authentifizierungsfehler".

Seriennumer 12x verglichen, leider kein Erfolg.
Gibt es eine maximal Anzahl an Wechselrichtern?
Meine 2x HMS800 funktionieren einwandfrei, aber bei dem neuen HM400 WR hapert es nun.
Im Readme kann ich nichts zu maximaler Anzahl finden...

Bilder lade ich sofort hoch!

SebiWolze, 2025-04-03:
[image]

[image]

[image]

gone-for-coding, 2025-04-03:
Das sieht so aus, als hättest du versucht ein Kommando zu senden. Hast du die Seiten mit einem Passwort geschützt? Dann musst du dich erst einloggen, bevor du Befehle senden kannst auch wenn man die Live Seite so sehen kann.

Daten kommen aber an?

SebiWolze, 2025-04-03:
> Das sieht so aus, als hättest du versucht ein Kommando zu senden. Hast du die Seiten mit einem Passwort geschützt? Dann musst du dich erst einloggen, bevor du Befehle senden kannst auch wenn man die Live Seite so sehen kann.
> 
> Daten kommen aber an?

Hey, nein, es wird nichts vom WR ausgelesen.
Ich habe versucht "Neustarten" zu senden, das klappte nicht.
Strom wird produziert, habe meine beiden anderen Wechselrichter vorhin kurz abgeschaltet und im KNX Stromzähler geschaut, da kam Stromproduktion von "Einzelplatte" an.
Ich sehe jedoch in AhoyDTU nichts, als wenn die Kommunikation scheitert.

gone-for-coding, 2025-04-03:
Andere Frage: Du hast an deinem ESP ein CMT-Modul für die HM**S**-800. Hast du für den HM-400 ein NRF-Modul und auch aktiviert?

mroenne2022, 2025-04-03:
@gone-for-coding : Ich denke auch das NRF Modul ist womöglich nicht korrekt oder garnicht aktiviert (oder nicht vorhanden ? falls es sowas gibt)

SebiWolze, 2025-04-03:
Gerade geschaut, die HMS 800 laufen auf 860Mhz und der 400er auf 2,4Ghz? Oh nein....
Welches 2. Modul benötige ich?

SebiWolze, 2025-04-03:
Okay, da liegt mein Problem.. habe nicht drauf geachtet und brauche das NRF Modul.
https://shop.blinkyparts.com/de/Ebyte-Funkmodul-NRF24L01-E01-ML01DP5-2-4-GHz-inkl.-passender-3dBi-Antenne/blink238942
Ordere dann das.
An den selben ESP?
Habe ESP32 in rauen Mengen liegen, überlege ein 2. AhoyDTU damit aufzubauen oder Quatsch?

mroenne2022, 2025-04-03:
@SebiWolze : Du brauchst ein NRF24L01+ Modul , das + ist wichtig! Und ja , du kannst natürlich einfach eine 2.DTU bauen mit dem NRF, die liest dann nur den HM400 aus.
Es geht aber auch alles an einer DTU

SebiWolze, 2025-04-03:
https://shop.blinkyparts.com/de/Ebyte-Funkmodul-NRF24L01-E01-ML01DP5-2-4-GHz-inkl.-passender-3dBi-Antenne/blink238942

Das hier ist ohne "+" -> wird aber direkt auf dieser Seite verlinkt: https://ahoydtu.de/getting_started/
Also das Modul taugt nicht?

> [@SebiWolze](https://github.com/SebiWolze) : Du brauchst ein NRF24L01+ Modul , das + ist wichtig! Und ja , du kannst natürlich einfach eine 2.DTU bauen mit dem NRF, die liest dann nur den HM400 aus. Es geht aber auch alles an einer DTU

SebiWolze, 2025-04-03:
[image]
Habe so ein NRF Modul noch hier liegen, ob es ein "+" Modell ist weiß ich nicht. Platine ist unbeschriftet.
Teste ich damit.

mroenne2022, 2025-04-03:
Das ist das richtige, das ist ein +  , auf dem quadratischen schwarzen Chip lesbar (zoomen)

mroenne2022, 2025-04-03:
[image]

SebiWolze, 2025-04-03:
> [image]

Satan, gut gesehen!
Ja sauber, Danke Euch!
Da war ich ja total auf dem Holzweg.
verdrahtet hab ich gerade, ESP flasht, ich mach einen 2. DTU damit fertig!

Merci!!!!

Ich schließe dann mal!

closed 2025-04-03 by SebiWolze