lumapu/ahoy #1824: ESP8266 nach Update auf v0.8.155 nicht möglich

RalfHS, 2025-04-28:
Nach Update auf die Version 0.8.155 kann der Webserver nicht mehr erreicht werden. 

Der ESP8266 bekommt von der Fritz!Box die IP zugewiesen, aber beim Aufruf der IP im Browser wird keine Webseite angezeigt.

Nach Rücksetzen über den angebotenen Webinstaller funktioniert alles wieder. Wenn jedoch das Update noch einmal versucht wird, dann schlägt es wieder fehl.

knickohr, 2025-04-28:
Der Update auf die 155 ist nicht zwingend notwendig, sondern bereitet nur den Versionswechsel auf die neue 0.9 vor. Da Du einen ESP8266 hast, wird der Versionswechsel auf die neue Ahoy nicht mehr supportet.

Bleib auf der alten funktionierenden Version.

Gubi2023, 2025-04-28:
komisch, bei mir funzt es wunderbar....

RalfHS, 2025-05-01:
Ich habe mich jetzt vom neusten Update 0.8.153 kommend, mit jeweiligem WebInstall der ganz alten Version, bis auf Version 0.8.151 zurück gearbeitet.

Die Version 0.8.151 ist die letzte Version, die auf dem ESP8266 funktioniert.

ricoshe81x, 2025-05-01:
kann man den auf einen neuen ESP32 migrieren? Ich hab ein ähnliches Problem mit meinem ESP8266 :(