lumapu/ahoy #1826: Update lässt sich nicht installieren...

Katerle, 2025-04-29:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.153

### Github Hash

5120aa4

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT deaktiviert

### Debug Serial Log output

```shell

```

### Error description

Ich habe aktuell die DEV-Version 250305_ahoy_0.8.153_5120aa4_esp8266-de.bin drauf.
Jetzt wollte ich ein Update auf die neue 155 machen. 250309_ahoy_0.8.155_5feb293_esp8266-de.bin
Datei auswählen, Aktualisieren, dann schreibt er "Aktualisierung gestartet" und das war's. Er macht sonst nix, hängt nicht, ich kann ganz normal weiter zugreifen. Das gleiche mit Version 154. Vom Strom getrennt, x-mal neugestartet, kein Ergebnis.
Ich habe im Lauf der Zeit bestimmt schon an die 10 Updates gemacht, ohne Probleme. Seit dem auch nix verändert.
Ratlosigkeit macht sich breit...
Könnt ihr helfen?
LG und Danke, Stefan

Gubi2023, 2025-04-29:
geh mal auf die .152 zurück und dann auf die .155 Die .153 ist evtl. buggy

Katerle, 2025-04-29:
Wollte ich, ging aber auch nicht. Trotzdem vielen Dank.
Ich hab's aber jetzt rausgefunden: Anderer PC, gleicher Browser: Firefox, gleiches Problem.
Dann mit M$-Edge probiert und es ging direkt auf die 155er...
Also, es liegt an FF.
Vielen Dank euch...

zeroboioioy, 2025-05-12:
Kann es bestätigen: In Firefox geht es nicht, in Chrome geht es.

Gubi2023, 2025-05-12:
dann hast du bei Firefox irgendwelche webblocker aktiviert. Bei mir geht es auch mit Firefox