lumapu/ahoy #1829: Wrong power values for HERF 800

fm94, 2025-05-17:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.155

### Github Hash

5feb293

### Build & Flash Method

ESP Tools (flash)

### Setup

- Configured the inverter to HERF800 using the serial number
- configured both solar panels (the inverter has 2 entries) with their max power (410W)

### Debug Serial Log output

```shell

```

### Error description

When I go to the home page, I see the total power which seems correct (reaches 800W) however, the DC Power of panel 0 seems to be wrong, e.g.:
Panel 0 100W, Panel 1 400W, Total power 800W.
Same thing for other values of Panel 0 except the daily yield which seems to be correct because it matches the sum.

Gubi2023, 2025-05-18:
what happens when you change Panel 0 with Panel 1? Same result?

fm94, 2025-05-18:
even when swapping, same results.
here's a screenshot of what I get

[image]

The sum of DC Power is wrong, but the daily yield is correct for example.
Also the Voltage and Current values are weirdly different for both panels

Gubi2023, 2025-05-18:
i dont see your Problem. The generated power per panel depends on the solar irradiation. The total power output is correct.

[image]

fm94, 2025-05-18:
But something is weird no? The total yield if each panel is almost the same but the DC Power is almost always the half. Also the efficiency is 140% which can't be theoretically.
I have the same panels which the same specs and no shades.

spies912, 2025-05-21:
Same issue here with HERF800 inverter

[image]

DC Power of channel 1 looks odd

fm94, 2025-05-21:
Yes, also the voltage and current of Channel 2 look very hight. My guess is that it's a problem with decoding the values received from the inverter.

Gubi2023, 2025-05-21:
i remember, that there ws a Issue, that HERF-Inverters count wrong, but i did`nt find it 🤔

spies912, 2025-05-22:
With byte assignment from opendtu it works fine for me

stefan123t, 2025-05-23:
@spies912 can you try to build the PR I just commited based on your observations and report back if this solves the issue for you ?
I have also updated the byte order in AhoyDTU from the OpenDTU HERF_1CH.cpp as the projects differed there too.

spies912, 2025-05-23:
> [@spies912](https://github.com/spies912) can you try to build the PR I just commited based on your observations and report back if this solves the issue for you ? I have also updated the byte order in AhoyDTU from the OpenDTU HERF_1CH.cpp as the projects differed there too.

This would also affect non-herf devices I guess. Is this intended?

stefan123t, 2025-05-23:
Good point 🙊, then @lumapu would need to introduce a distinction between HERF and Hoymiles devices to make this work.

fm94, 2025-05-23:
There's already this PR https://github.com/lumapu/ahoy/pull/1834

I think it's a good candidate. I'll test it.