lumapu/ahoy #1830: HMS-2250 6T - Switch to "power limits per line"

AlexanderMaydorf, 2025-05-19:
Hi,

With the original Hoymiles DTU, I'm able to set the 2250 6-T to power limit mode per individual line.
However, even in the latest version of AhoyDTU, it seems there's only a global limit available — which is unfortunately quite limiting in practice.

Is there an update planned to support line-specific power limits? Or is this feature already on the roadmap?

Best regards,
Alexander

## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

ms49434, 2025-05-19:
Please review your issue. There's no model HMS-2250 6-T, you are probably running a HMT-2250-6T.
HMT-1800-6T and HMT-2250-6T cannot be limited per trunk, they can be limited only globally.
At the time of writing, there's no known firmware which would support PDL on these models.

AlexanderMaydorf, 2025-05-19:
You are right, I got a wrong information.

closed 2025-05-19 by AlexanderMaydorf