lumapu/ahoy #1839: wie schreibe ich auf die AhoyDTU?

AugustQu, 2025-06-04:
Hi,

von der AhoyDTU kann ich lesen, aber wie schreibe ich einen Wert auf diese?

Mit einem Python-Skript kann ich Daten von der AhoyDTU auslesen. Das Skript läuft auf einem RasPi, via cron.

In der Oberfläche der AhoyDTU sehe ich unter Serial/Control einen Eintrag: Send Power Limit

Wie kann ich das per Programm machen? Ich will diese Einstellung nicht per Oberfläche sondern über mein Python-Skript vornehmen.

Any tips?
Danke
AugustQ

PS: meine Anlage:
2x Panels a 400W
WR: Hoymiles HM 600
AhoyDTU in der Version 0.6.9 (sollte mal upgraden)
mit ESP8266
weitere Infos benötigt?

AugustQu, 2025-06-04:
Danke. Ich sehe mir das an.

PS: diese Doku hatte ich bisher nicht gefunden, deshalb meine Frage.

knickohr, 2025-06-04:
👍